badgeek/svg2shenzhen #1: Prepare document keeps popping up the popup

kasbah, 2018-03-30:
When I press "prepare document" and then click apply the popup re-appears. I have to click cancel to stop it.

badgeek, 2018-03-30:
hi kaspar, did you get export function working?

5shekel, 2018-03-31:
might be connected, 
python2 was missing lib2xml 

but it did produce a clear error message
https://i.imgur.com/23z0S1a.png

badgeek, 2018-04-01:
just hit close after "prepare document"

kasbah, 2018-04-01:
So this is how it's supposed to be? Why not close it automatically after preparation completes? Seems confusing to do it this way. 

> did you get export function working?

Yes, I did manage to get the export working in the latest version. This is going to be awesome when I have more time to play with it!

badgeek, 2018-04-01:
yes you have to hit close, i tried to find why this behaviour is happening but have no clue at all.. been loking other extension as well...

kasbah, 2018-04-03:
I feel like it is a bug in that case and should remain open until a solution is found. I'll have a go at solving it (when I find time).

badgeek, 2018-04-03:
kool, thx

kasbah, 2018-04-13:
I had a look over the inkscape docs I could find but haven't found a way to automatically close it either. :frowning_face:

closed 2020-08-17 by badgeek