badgeek/svg2shenzhen #2: linux

5shekel, 2018-03-31:
following instructions. working with example virusPCB svg
after 
`4.Extension > Svg2Shenzhen > Export Kicad`
i get 

1. png files in output folder (they look fine)
2. this error popup
`"/home/yair/.config/inkscape/extensions/bitmap2component_linux64" "/home/yair/dev/testPCB/001_F.Cu.png"
"/home/yair/dev/testPCB/001_F.Cu.kicad_module" F.Cu true 128
"/home/yair/.config/inkscape/extensions/bitmap2component_linux64" "/home/yair/dev/testPCB/002_F.Mask.png"
"/home/yair/dev/testPCB/002_F.Mask.kicad_module" F.Mask true 128
"/home/yair/.config/inkscape/extensions/bitmap2component_linux64" "/home/yair/dev/testPCB/003_F.Silk.png"
"/home/yair/dev/testPCB/003_F.Silk.kicad_module" F.Silk true 128
Traceback (most recent call last):
  File "export_layers.py", line 493, in <module>
    _main()
  File "export_layers.py", line 489, in _main
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
    self.effect()
  File "export_layers.py", line 228, in effect
    self.processExportLayer()
  File "export_layers.py", line 278, in processExportLayer
    with open(kicad_file, 'r') as myfile:
IOError: [Errno 2] No such file or directory: '/home/yair/dev/testPCB/001_F.Cu.kicad_module'
`
archLinux

badgeek, 2018-03-31:
Could you paste the result if you run this in terminal

/home/yair/.config/inkscape/extensions/bitmap2component_linux64

5shekel, 2018-03-31:
```
% ls
001_F.Cu.png          bitmap2component_linux64  export_layers.inx  prepare_doc.inx  svg2shenz
bitmap2component.exe  bitmap2component_osx      export_layers.py   prepare_doc.py

% sh bitmap2component_linux64 001_F.Cu.png
bitmap2component_linux64: bitmap2component_linux64: cannot execute binary file

% chmod +x bitmap2component_linux64 

% file bitmap2component_linux64                                                 :(
bitmap2component_linux64: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=4fe444aaadf037158cf3d1be4199d6e620cc111c, not stripped
```

5shekel, 2018-03-31:
```
% ./bitmap2component_linux64 001_F.Cu.png 

 bitmap2component-cli 

 Usage
 ./bitmap2component <filename.png> <output filename> <layer> <negative|true/false> <threshold|0-255>

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
[1]    8811 abort (core dumped)  ./bitmap2component_linux64 001_F.Cu.png
```

5shekel, 2018-03-31:
so it works...
```
 % ./bitmap2component_linux64 001_F.Cu.png output                                :(

 bitmap2component-cli 

 Usage
 ./bitmap2component <filename.png> <output filename> <layer> <negative|true/false> <threshold|0-255>

[bitmap2component] Filename 001_F.Cu.png
[bitmap2component] Open bitmap
[bitmap2component] Loaded png s: 2362 x 2362
[bitmap2component] Create bitmap
[bitmap2component] Fill bitmap
[bitmap2component] Trace image
[bitmap2component] Done
```

5shekel, 2018-03-31:
i guess it was the 
`chmod +x bitmap2component_linux64`
that solved it

badgeek, 2018-03-31:
cool, i think i lost +x when i transfer it from linux to osx, i'll update the zip

badgeek, 2018-03-31:
meanwhile you can try to export from inkscape it should work

5shekel, 2018-03-31:
while your doing the zipping, delete the __MACOS folder.

badgeek, 2018-03-31:
coool,,, updated latest build file

5shekel, 2018-03-31:
looking good
https://i.imgur.com/v33FFAQ.png

badgeek, 2018-03-31:
looks like it missing drill pad and silk screen ?

closed 2018-03-31 by 5shekel

5shekel, 2018-03-31:
no drills and screens?
just me opening footpads and not the compiled.kicad_pcb file

alexpacini, 2019-10-27:
I still have the same issue in the latest version of the zip (0.2.17).
Maybe you need to use a tar.gz to keep linux file permissions.