badgeek/svg2shenzhen #4: rounded corners on boxes

5shekel, 2018-03-31:
might be the small size of the sketch?  
all objects are converted to a path.  
but i get rounded corners where there shouldn't be  

[image]

[svg srouce](https://github.com/shenkarSElab/TinyTony/blob/master/board/TinyTony_ShenzenReady_board.svg)

badgeek, 2018-03-31:
image is too small, fix set export dpi to 2000 but the extension that you have now max dpi is 1000... will add the fix on next release

[image]

5shekel, 2018-03-31:
my setup limits me to 1000dpi, even if i set it on 2000.
after i updated to your fix

badgeek, 2018-03-31:
aha i forgot to update readme, its now pointing to release section

badgeek, 2018-04-01:
bitmap2component will still import using 600 dpi as scale, fixed in version 0.2.3