badgeek/svg2shenzhen #5: Not creating files in Windows

twinkletwinkie, 2018-04-08:
Attempting to run the Export command gives errors:

`IOError: [Errno 2] No such file or directory: 'C:\\Users\\username\\pcbart.pretty\\001_F.Cu.kicad_mod'`

Even if I manually create the directory it's looking for it won't create the files.  It appears to work fine when done in Ubuntu.

twinkletwinkie, 2018-04-08:
Folders:
Pcbart.pretty
Images

Images has .png files.

twinkletwinkie, 2018-04-08:
[image]

badgeek, 2018-04-09:
extract and open bitmap2component, and what is the output?

[bitmap2component.exe.zip](https://github.com/badgeek/svg2shenzhen-next/files/1888060/bitmap2component.exe.zip)

twinkletwinkie, 2018-04-09:
[image]

badgeek, 2018-04-09:
no bitmap2component.exe ?

badgeek, 2018-04-09:
just run the exe./..

twinkletwinkie, 2018-04-09:
I misunderstood what you were asking me. I didn't realize that you had linked to an updated version of bitmap2component.exe.  It appears to be working correctly now, thank you very much.

twinkletwinkie, 2018-04-09:
I processed the example you provided and it says "LOGO***" in the middle of the board.  Is it supposed to do that?

[image]

badgeek, 2018-04-09:
yes that is correct, i will push new release with this fix

closed 2018-04-09 by badgeek