badgeek/svg2shenzhen #7: Bitmap2component compilation fails at first step

kasbah, 2018-04-12:
```
$ g++ -c ../common/geometry/* -lpotrace -L. -L../potrace -I../include -I../polygon -I../potrace -I.  -std=c++11 
../common/geometry/hetriang.cpp:50:40: fatal error: class_board_connected_item.h: No such file or directory
compilation terminated
```

badgeek, 2018-04-12:
Are you using kicad 4.0.7 source?

kasbah, 2018-04-12:
No, I thought this was self contained. I need to put the source somewhere in particular?

badgeek, 2018-04-12:
Aha, i forgot abt this, kicad source already included but you need lib
boost  (im using ver 1.56) put it inside include/boost folder

badgeek, 2018-09-21:
download boost here 
https://github.com/badgeek/svg2shenzhen/files/2404261/boost.zip

badgeek, 2018-09-21:
compilation step is in #31

kasbah, 2018-09-21:
It will take a me a few days at least to get around to doing this. Feel
free to go ahead if it's urgent.

badgeek, 2018-09-21:
no problem... i'll try to document it while i have free time...

closed 2020-08-17 by badgeek