badgeek/svg2shenzhen #11: zone support

badgeek, 2018-04-20:
create kicad zone will be good addition for this extension

blark, 2020-01-23:
Agree 100% if you use a lot of copper on one side of the PCB you're currently severely limited in vias which makes even moderately complex designs hard or impossible to do. If shapes were filled zones that you could run a via into it would be really nice.

Colecaog, 2021-05-12:
I would also like this.  Right now my workaround (suggested by someone on the KiCAD facebook group) is to create a zone on the board and close the board file, open up the .mod file for the copper in question, copy the pts code from the polygon, open the board file in a text editor and find the zone and paste the points file from the copper polygon.  It works but is a bunch of extra work, I would think it should be easy to change from fp_poly to zone in the plugin?