badgeek/svg2shenzhen #12: Use variables in output path

kasbah, 2018-04-20:
Instead of setting output path to a folder `~/output` we could set it to the exact filename `~/output/my.kicad_pcb` but also allow filename variable to be substituted. 

E.g. `~/output/{filename}` could expand to:

-  `~/output/drawing.kicad_pcb` if the svg is called `drawing.svg` and the output is set to kicad_pcb (it would still also create a `drawing.pro` and `fp-lib-table`)
-  `~/output/edge_conector.kicad_mod` if the svg is called `edge_connector.svg` and the output filetype is set to kicad_mod

If the output is set to folder we could use the old behaviour. Thoughts?

kasbah, 2018-04-20:
Or maybe we could just change the default behaviour and always use the svg filename as the output name? Might be less confusing than using python `format` style variable substitution.

badgeek, 2018-04-20:
Been thinking abt this too, bit problem is we cant get current svg filename
from inkscape

kasbah, 2018-04-20:
I think this is good as it is now with merging of #13. No need to complicated things with variables. Feel free to re-open if you feel differently.

closed 2018-04-20 by kasbah