badgeek/svg2shenzhen #14: KiCad 5 anchor pad support

kasbah, 2018-04-20:
Noticed this in @Junes-PhD's [kicad_arbitrary_pad_example](https://github.com/Junes-PhD/kicad_arbitrary_pad_example). I missed the Hackchat but it looks like KiCad 5 will support funky pad shapes which means you can better make drawings part of the circuit without breaking the rules checks. This won't work with KiCad 4 of course. 
[image]

```lisp
  (pad 1 smd custom (at 0 0) (size 0.001 0.001) (layers F.Cu F.Paste F.Mask)
    (zone_connect 0)
    (options (clearance outline) (anchor rect))
    (primitives
      (gr_poly (pts
         (xy -12.942245 10.707266) (xy -12.253724 11.874456) (xy -11.425987 13.074738) (xy -10.645207 14.040135) (xy -9.361396 15.444829)
         (xy -11.699119 17.536869) (xy -12.619469 18.358458) (xy -13.382433 19.035699) (xy -13.906951 19.496904) (xy -14.111184 19.670024)
         (xy -14.162957 19.935938) (xy -14.225695 20.590957) (xy -14.28255 21.439588) (xy -14.639823 23.402138) (xy -15.43931 25.1571)
         (xy -16.617038 26.643019) (xy -18.109037 27.79844) (xy -19.851333 28.561908) (xy -21.779955 28.871969) (xy -22.023363 28.875789)
         (xy -22.857384 28.840815) (xy -23.460461 28.751177) (xy -23.63843 28.676657) (xy -23.531601 28.430362) (xy -23.094161 27.905675)
         (xy -22.395586 27.178997) (xy -21.505349 26.326727) (xy -21.481343 26.304569) (xy -19.125125 24.131612) (xy -20.725194 22.337724)
         (xy -21.619116 21.334803) (xy -22.537778 20.302963) (xy -23.308394 19.4363) (xy -23.431939 19.297174) (xy -24.538614 18.050513)
         (xy -26.506676 19.768703) (xy -27.4354 20.579181) (xy -28.280475 21.316071) (xy -28.914224 21.868069) (xy -29.111737 22.039762)
         (xy -29.748738 22.592631) (xy -29.612724 20.816598) (xy -29.217555 18.763899) (xy -28.39627 17.004342) (xy -27.192862 15.576228)
         (xy -25.651325 14.517862) (xy -23.81565 13.867548) (xy -21.729831 13.663588) (xy -21.408508 13.673374) (xy -19.56924 13.758343)
         (xy -17.337778 11.775775) (xy -16.371865 10.917968) (xy -15.500868 10.145121) (xy -14.834812 9.554823) (xy -14.530894 9.286149)
         (xy -13.955472 8.779092) (xy -12.942245 10.707266)) (width 0))
    ))
```

badgeek, 2018-04-20:
i missed hackchat too.. 6 am here... this would be good addition

badgeek, 2018-04-20:
in kicad 4 maybe we can use zone

kasbah, 2018-04-20:
Found a [transcript of the Hackchat](https://hackaday.io/event/112119-pcb-artwork-and-photo-conversions/log/144727-pcb-artwork-photo-conversions-hack-chat-transcript). Some other good tips in there too.

Sodium-Hydrogen, 2020-10-06:
I just opened a pull adding this #105