badgeek/svg2shenzhen #22: Please release 0.3

kasbah, 2018-05-13:
Or 0.2.13, not sure how the version numbers work. I am holding an [svg2shenzhen workshop in Paris](https://hackarnaval.online/abstracts-fr/kicad.html) this week and it would be nice to have easy access to the latest merged features.

badgeek, 2018-05-13:
Hi Kaspar, i just make 0.2.13 release
https://github.com/badgeek/svg2shenzhen/releases

im thinking of about making svg2shenzhen inside organization repo so you
can also make new release without my approval

im bit slow since im working on another project and havent start making pcb

what do you think?

cheers

kasbah, 2018-05-13:
Thanks! You mean make a GitHub org and add me to it? Sure, you could also just add me to this repo I think. If I were to make a future release I would need to know: is there some thinking behind the version numbers or is it not important?

badgeek, 2018-05-14:
cool, version number is not important, just to make each release have different number (im reaaly poor about using good convention) if you have better version number, go for it, and my consideration for each release is compatibility with windows  (make sure windows version also working well), i'll think abt organization name or maybe i'll add you to this repo (i dont know that before)

closed 2018-05-20 by kasbah