badgeek/svg2shenzhen #24: via annular ring

5shekel, 2018-05-19:
reproduce:  
* create a (0.6mm) circle in drill layer   
* a 1mm copr ring around it is generated automatically in copper layer.

kasbah, 2018-05-20:
I can't reproduce this. Is svg2shenzhen up to date? If it is, can you share the file where this happens?

kasbah, 2018-05-20:
Oops, actually I can reproduce this: I can see it in Gerbers but not in KiCad. I guess KiCad adds the ring. The solution is probably to mark these holes NPTH (non-plated through-hole) but then they will also be put into the NPTH.drl file unless you check an option to merge them. Still makes the most sense since they could in fact be NPTH.

5shekel, 2018-05-20:
so to clarify, 
the diff between a NPTH and thru hole is
**Thru hole:** 
the script adds ring in mask and copper layer (front and back) to all drill holes, **by default**  

**NPTH:** 2 post script options
1. edit hole in kicad, so its specified as NPTH or Thru Hole(by default)  
2. remove the ring from all affected layers in gerber editor (gerbv) which will basically turn it into an NPTH  

btw. the source is [here](https://github.com/shenkarSElab/candy/blob/master/gerber/gfx/thermoCandy-svg2shnzn.svg)   
using release (0.2.13)

kasbah, 2018-05-20:
By "the script" you mean svg2shenzhen? Currently it adds though-hole pads with the copper and mask the same size as the hole. So yeah, your two workarounds should work. We should fix this in svg2shenzhen though.

5shekel, 2018-05-21:
@kasbah can you direct me to the location of the drill creation in the svg2shenzhen script?  
i want to experiment and a bit lost .

kasbah, 2018-05-21:
https://github.com/badgeek/svg2shenzhen/blob/fb80c3ac2de66e3bdf15c447a4b5d6e00d70a3d1/inkscape/svg2shenzhen/export.py#L619

I think we can change those to `np_thru_hole` rather than `thru_hole`. I believe we can also remove the pad names since it makes even less sense to give them names. 

I played around a bit and noticed there is still a solder mask expansion around the non-plated holes. I believe that may be needed anyway.

4b11b4, 2018-10-26:
> I played around a bit and noticed there is still a solder mask expansion around the non-plated holes. I believe that may be needed anyway.

Where did you see the solder mask expansion? In the exported gerbers?
Just wondering if it might be due to local or global mask clearances that would not be due to svg2shenzen.

5shekel, 2018-10-26:
yes, its an auto expansion done by kicad to all thru holes. 

you can choose to disable that expansion inside kicad.  by right click/editing each pad before you export to gerber (maybe even change the default template in kicad, idk)

i use a kicad script (#25) to bypass the kicad GUI stage altogether so i erase any kicad "addons" i dont like directly at the gerber stage. using gerber viewer and editor(!) > [gerbv](gerbv.sourceforge.net) <3

drayde, 2019-11-03:
I changed export.py to support an optional Drill-NP layer, that generates non-plated holes. 
In case you think this is useful not just for me, check out the PR I created: https://github.com/badgeek/svg2shenzhen/pull/55