badgeek/svg2shenzhen #30: 0.2.13 Plugin No Longer Runs

Tom-Archer, 2018-08-09:
The zip file, "svg2shenzhen-extension.zip" that makes up release 0.2.13 contains the following code in "export.py":

`    def processExportLayer(self):
        insertion_date = dateutil.parser.parse('2018-07-28 14:25:30')
        time_between_insertion = datetime.now() - insertion_date

        if (time_between_insertion.days>7):
            return`

This now means the plugin doesn't run. Was this planned obsolescence?

Tom-Archer, 2018-08-09:
This also places a dependency on the "dateutil" package, which isn't imported at the top of the file, nor is it installed by default as part of a Python installation. Took me a bit of hacking to get this script to actually run.

kasbah, 2018-08-12:
Can confirm that's what's in the zip. I have been using the git version so never checked the zip. @badgeek, what the hell!? I didn't consent to this! Pretty easy to remove but, just, why? There needs to be a release with a fix ASAP and an explanation!

Here is a fixed zip in the meantime: https://cloud.monostable.co.uk/svg2shenzhen-fb80c3a.zip.
Someone can verify that this is a straight up zip of the Inkscape folder at state of the code at the fb80c3a commit for reassurance. 

There are compiled binaries in the source tree though. Are those modified from the source as well @badgeek? Will they do some stupid shit too!?

kasbah, 2018-08-12:
I just realised I have the ability to make releases so I made https://github.com/badgeek/svg2shenzhen/releases/tag/v0.2.14. I deleted the zip for v0.2.13. I still need an explanation for this nonsense though.

badgeek, 2018-08-12:
Thanks i will explainnsoon