badgeek/svg2shenzhen #32: PCBs made using svg2shenzen

badgeek, 2018-09-21:
would be cool if we have list/gallery of pcb made with this tool

5shekel, 2018-10-08:
i found it allows me to design footprints out of this world 
like this hi-rose connector adapter from a screw type to board

i will look around

4b11b4, 2018-10-18:
I have a bunch of faceplates that I made for analog synthesizers.

Tom-Archer, 2018-10-22:
Here's one of mine:
https://www.tindie.com/products/tomm_archer/myah-badge-inspired-by-skeletor/
http://www.tomarcher.co.uk/2018/08/29/pcb-pin-badges-skeletor/

4b11b4, 2018-10-29:
https://ibb.co/gvrKWV

badgeek, 2018-10-29:
> https://ibb.co/gvrKWV

do you have actual photo of the board?

kasbah, 2018-10-29:
New Kitspace ruler is looking much better. https://kitspace.org/boards/github.com/kitspace/ruler/

4b11b4, 2018-10-29:
> > https://ibb.co/gvrKWV
> 
> do you have actual photo of the board?

I have a variety of boards like this, I'll get some photos

5shekel, 2018-10-29:
https://github.com/5shekel/pdRuler

ssilverm, 2019-02-10:
https://imgur.com/a/zQlW1Rs

https://blog.adafruit.com/2019/02/04/the-making-of-pcb-art-pcb-art-making-badgelife-ssilver2k2/

badgeek, 2019-02-10:
thanks, i'll post what i found as well,

https://www.gabetaubman.com/blog/posts/kicad-custom-footprint/

Tom-Archer, 2019-02-10:
Here's another one of mine:
http://www.tomarcher.co.uk/2018/12/10/xmas-fireplace-ornament-pcb/

5shekel, 2019-02-15:
fresh... noticed (while googling svg2sz)
https://www.seeedstudio.com/blog/2019/02/14/new-year-lions-another-attempt-at-kicad-svg2sz-pcb-art/

badgeek, 2019-02-20:
http://dernulleffekt.de/doku.php?id=paper_pcb_shenzhen_ready:main

Tom-Archer, 2020-08-18:
Here's another couple of mine: http://tomarcher.co.uk/2019/11/16/pcb-pin-badges-roberto/
http://tomarcher.co.uk/2020/05/16/pcb-pin-badges-clamps-sao/

You've got two of my Skeletor badges on the front page image so I don't mind if you remove the non-illuminated one 😉