badgeek/svg2shenzhen #35: Checkbox to hide Val and Ref

4b11b4, 2018-10-15:
I was wondering if it might be possible to add a checkbox to set the Value and Reference text to "not visible" from the Step 2 Menu within Inkscape?

kasbah, 2018-10-16:
You mean when you export the whole thing to a kicad module? I think we should just make it invisible by default.

4b11b4, 2018-10-16:
I would agree.
Most of the time you'll be making features which won't need the Val and Ref silkscreen fields?
A checkbox (checked by default) would give people the option though.

kasbah, 2018-10-17:
I think we don't even need a checkbox because you can fairly easily override/change this in KiCad.

4b11b4, 2018-10-18:
True. It is probably the default case that people will not need a %V and %R, since this is for making symbols and art.

4b11b4, 2018-10-26:
@badgeek What do you think? I will make a pull request.

4b11b4, 2018-10-26:
This would be appended after line 463:

(fp_text reference Ref** (at 0 0) (layer F.SilkS) hide
    (effects (font (size 1.27 1.27) (thickness 0.15)))
  )
  (fp_text value Val** (at 0 0) (layer F.SilkS) hide
    (effects (font (size 1.27 1.27) (thickness 0.15)))
  )

4b11b4, 2018-10-26:
#37

badgeek, 2018-10-27:
i'll test it

closed 2018-10-29 by 4b11b4