badgeek/svg2shenzhen #38: Unable to export .Mask (MacOS)

vektorious, 2018-12-07:
First of all, thanks for this great Inkscape addon! 

I am playing around with it since a few days and it works great for exporting F.Cu and Edge.Cuts but I run into problems when I wanted to include a F.Mask layer.

I renamed the layer and used a copy of my F.Cu drawing and also tried F.Mask-auto but neither works for me. It seems like only F.Cu and Edge.Cuts always are exported. I still have to check if F.Silk works because then I could do the texteditor rename workaround.

I am not sure if it's a OS problem but Inkscape does not run as smoothly on MacOS than it does on  Linux/Windows (from my experience). [Here are all my files](https://gitlab.com/alexander.kutschera/microbadge/tree/master/pcb)

badgeek, 2018-12-07:
hi, F.Mask should be work, i'll check it on weekend

badgeek, 2018-12-07:
what is your inkscape version?

badgeek, 2018-12-07:
@vektorious interesting pcb! btw check this, 

[bacterialcell.zip](https://github.com/badgeek/svg2shenzhen/files/2656535/bacterialcell.zip)

i think fmask is working

vektorious, 2018-12-07:
It's 0.92.2 5c3e80d from 2017-08-06 so fairly up to date.
I think atm the latest stable version is 0.92.3

I'll be around on Saturday so if you want me to quickly test s.th let me know!

vektorious, 2018-12-07:
> @vektorious interesting pcb! btw check this,
> 
> [bacterialcell.zip](https://github.com/badgeek/svg2shenzhen/files/2656535/bacterialcell.zip)
> 
> i think fmask is working

Oh wow. I am super sorry. I feel stupid now. You are right I just did it again and it is working with copying the F.Cu sketch to F.Mask and then exporting it.

I don't know what went wrong earlier because I also checkt the 3D viewer and everything... 

Sorry again, thanks for the quick help!

@kasbah introduced me to PCB design and now I am tinkering around with this design which basically is (or will be) very similar to @dusjagr [Gär Lämpli](https://www.hackteria.org/wiki/G%C3%A4r_L%C3%A4mpli) for indicating the right temperature for bacteria growth. 

I still have to learn a lot though.

closed 2018-12-07 by badgeek