badgeek/svg2shenzhen #44: Unable to get anything exported…

hedefalk, 2019-05-05:
I'm trying to export an svg, but I guess I'm missing some magic sauce because I just keep getting this empty footprint:

```
(module "test" (layer F.Cu)
(fp_text reference Ref** (at 0 0) (layer F.SilkS) hide
(effects (font (size 1.27 1.27) (thickness 0.15)))
)
(fp_text value Val** (at 0 0) (layer F.SilkS) hide
(effects (font (size 1.27 1.27) (thickness 0.15)))
)
)
```

This is my svg:

https://gist.github.com/hedefalk/5b428772f7deefc906a194f297371e9e

hedefalk, 2019-05-05:
I'm on OSX 10.14.4 (18E226) and Inkscape 0.92.4 (5da689c313, 2019-01-14).

Just got feedback from other thread over at kicad forum that this file went fine on Ubuntu 0.92.2:

https://forum.kicad.info/t/svg-import-tips/16790/10?u=hedefalk

Debug output is:

```
inkscape C -d 600.0 -e "/Users/viktor/test/.svg2shenzhen-cache/images/F.Cu_g202.png" "/Users/viktor/test/.svg2shenzhen-cache/Q3Aisy"
inkscape C -d 600.0 -e "/Users/viktor/test/.svg2shenzhen-cache/images/F.Silk_g206.png" "/Users/viktor/test/.svg2shenzhen-cache/Rqyhms"
"/Users/viktor/.config/inkscape/extensions/svg2shenzhen/bitmap2component_osx" "/Users/viktor/test/.svg2shenzhen-cache/images/F.Cu_g202.png" "/Users/viktor/test/.svg2shenzhen-cache/test.pretty/F.Cu_g202.kicad_mod" F.Cu true 600 128
"/Users/viktor/.config/inkscape/extensions/svg2shenzhen/bitmap2component_osx" "/Users/viktor/test/.svg2shenzhen-cache/images/F.Silk_g206.png" "/Users/viktor/test/.svg2shenzhen-cache/test.pretty/F.Silk_g206.kicad_mod" F.Silk true 600 128
```

If I try to export png I get no file at all.

hedefalk, 2019-05-06:
I wonder if there's anything I'm missing. Read that it should be pitch black somewhere in a tutorial but other than that I don't know… I have very limited knowledge about Inkscape:

[image]

badgeek, 2019-05-06:
what is your inkscape version? there is nothing to do with svg format, all
layer exported to png by inkscape before going through bitmap2component...

badgeek, 2019-05-06:
@hedefalk can you try with inkscape 0.92.1, that is what im using and im on osx 10.13.6

hedefalk, 2019-05-07:
Ok, I didn't read carefully enough and tried with an osx installer 0.92.2. That one doesn't seem to work at all on my system. The Quartz process just pops up this broken window:

[image]

I'll try to give 0.92.1 a try!

hedefalk, 2019-05-07:
Actually, I don't know how to try Inkscape 0.92.1. The inkscape homepage referrs to this broken homebrew cask for that version which also actually is 0.92.2: https://github.com/caskformula/homebrew-caskformula

That cask is really old gives the following error:

```
Cloning into '/usr/local/Homebrew/Library/Taps/caskformula/homebrew-caskformula'...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 5 (delta 0), reused 3 (delta 0), pack-reused 0
Unpacking objects: 100% (5/5), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/caskformula/homebrew-caskformula/Formula/inkscape.rb
inkscape: "cxx11" is not a recognized standard
Error: Cannot tap caskformula/caskformula: invalid syntax in tap!
```

I'll try to see if I can come up with something. Feel free to close is you like and I'll  reopen if I can repro on other versions too… :)

badgeek, 2019-05-08:
@hedefalk try https://inkscape.org/pt/release/inkscape-0.92.2/mac-os-x/107/dmg/dl/

hedefalk, 2019-05-08:
@badgeek Wow, yes, there it works! Thanks!

closed 2020-08-17 by badgeek