badgeek/svg2shenzhen #45: Prepare document gives lxml dependency error (Windows).

ninjatill, 2019-05-16:
Hello,

I am using Windows 7 with Inkscape 0.92.4 and svg2shenzhen 0.2.16. When running prepare document, I'm getting an error that the "...lmxml wrapper for libxml2 is required...". I ran an administrator command prompt and ran "pip install lxml". The install finishes successfully. I close Inkscape and reopen and run prepare document again... I get the same dependency error. If I go back to the windows command prompt and run the "pip install lxml" command again, it says that the "requirement already satisfied". So then the question is how to get around this dependency error?

The other issues similar do not seem to have a specific resolution for this dependency error after the dependency is installed via pip. Thanks,

ninjatill, 2019-05-16:
Prepare docs dependency error:
[image]

After installing lxml via pip, if I run "pip install lxml" again, I get this message that the package is already installed:
[image]

kasbah, 2019-05-16:
It's likely that pip and inkscape are using different versions of python. I believe the plugin only works with python2 and I can see in your screenshot pip is working with python 3.6 (`python_36_64`). Unfortunately I don't use Windows very much so don't know how to resolve it.

ninjatill, 2019-05-17:
OK. That makes sense. I'll see if I can resolve that. I have multiple software products on the PC that require different versions.

closed 2020-08-17 by badgeek