badgeek/svg2shenzhen #46: bitmap2component is not buildable

onitake, 2019-06-17:
It is not possible to build bitmap2component with the shell scripts or cmake files provided with the source code.

It looks like the cmake files were taken directly from the KiCad source code, but they require a lot more build infrastructure and dependencies to compile cleanly.

When running `src/build.sh`, the following error is shown:
```
../common/geometry/hetriang.cpp:50:10: fatal error: class_board_connected_item.h: No such file or directory
 #include <class_board_connected_item.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

It's also unclear which version of KiCad these sources are based on.

Would it be possible to use the bitmap2component executable that is provided with KiCad instead? This would make it much easier to package svg2shenzhen without relying on precompiled binaries.

On Linux, distribution packages would install it to`/usr/bin/bitmap2component` (or another other location on `$PATH`). On OSX, the binary is in `/Applications/KiCad/KiCad/kicad.app/Contents/Applications/bitmap2component.app/Contents/MacOS/bitmap2component`. I don't know the exact location on Windows.

Alternatively, there's also this project: https://github.com/mtl/svg2mod
Converting vector graphics directly to outlines is cleaner than with an intermediate image step, and it would be possible to do everything in Python.

badgeek, 2019-06-17:
hi onitake, the reason that i custom build bitmap2component, the one from kicad does not accept command line parameter...

onitake, 2019-06-21:
Thanks for the clarification.
I still think it would be worthwhile to replace the patched bitmap2component with code from svg2mod. Converting vector graphics directly is way better than producing images and running them through potrace to obtain vector graphics again.

In the meantime, here's a PR that will at least allow building bitmap2component with the provided sources: #47

badgeek, 2019-06-23:
@onitake i disagree with this, i did this with the early version of svg2sz..it cause many problem and not as flexible as bitmap2component https://github.com/badgeek/svg2shenzhen-legacy

closed 2020-08-17 by badgeek