badgeek/svg2shenzhen #49: Svg2Shenzhen fails with Python errors in Inkscape 1.0alpha2

eckmensch, 2019-08-25:
You might be aware that the Inkscape team is currently actively working on the release of Inkscape 1.0. The team already published the second alpha to download. It looks like the Python extension API has slightly changed, so that the extension Svg2Shenzhen is not working any more.

#### Steps to reproduce:

1. Download Inkscape 1.0alpha2: https://inkscape.org/release/1.0alpha2/platforms/
2. Install Svg2Shenzhen in the user extension folder 
3. Open Inkscape 1.0alpha2 with a new document
4. Run svg2shenzhen using the menu `Extensions` > `Svg2Shenzhen SVGSZVER` (note the strange name extension) > `1. Prepare Document…`
5. Choose default parameters and click `Apply`

#### What happened?
The document was prepared properly. Size, layers and guides are set up, but the extension returns the following warnings:

```shell
prepare.py:24: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--docwidth", action="store", type="float", dest="docwidth", default=0.0)
prepare.py:25: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--docheight", action="store", type="float", dest="docheight", default=0.0)
prepare.py:295: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
prepare.py:164: DeprecationWarning: inkex.etree was removed, use "from lxml import etree"
  grid = inkex.etree.Element(inkex.addNS('grid','inkscape'))
```

#### What should have happened?
The document should have been prepared without error.

Judging by the warning messages, Svg2Shenzhen is using function calls, which a deprecated in Inkscape 1.0alpha2.

#### Continue the test with exporting to Kicad

6. Run svg2shenzhen using the menu `Extensions` > `Svg2Shenzhen SVGSZVER` (note the strange name extension) > `2. Export Kicad…`
7. Choose default parameters and click `Apply` white list area (see screenshot)

#### What happened?
The export file is not getting written to disk, and the extension returns the following warnings and errors:

```shell
export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:736: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:336: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 740, in <module>
    _main()
  File "export.py", line 736, in _main
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 111, in run
    self.save_raw(self.effect())
  File "export.py", line 313, in effect
    self.processExportLayer()
  File "export.py", line 337, in processExportLayer
    layers = self.get_layers(curfile)
  File "export.py", line 537, in get_layers
    if  layer_label_name in self.layer_map.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'
```

#### What should have happened?
The Kicad module should have been created without error.

This time there are depreciation warnings, but also errors, which seem to prohibit that Svg2Shenzhen succeeds.

#### Inkscape Version and Operating System:

- Inkscape Version: 1.0alpha2 (863a58d, 2019-08-17), official notarized build
- Operating System version: macOS Mojave 10.14.6 (18G87)

badgeek, 2019-08-26:
@eckmensch thanks for the detailed report, i will try to fix this on the weekend, cheers

eckmensch, 2019-08-26:
@badgeek This sounds great, thank you for your support!

I'm a bit involved in the testing of the Inkscape 1.0 development branch, preliminary focussed on the Mac version. I just came across an issue that discusses the planned, slight rework of the extension syntax, preliminary to simplify it and make preparations to allow for translation of individual extensions. Maybe this is interesting for you as well: https://gitlab.com/inkscape/inkscape/issues/333

badgeek, 2019-08-30:
> @badgeek This sounds great, thank you for your support!
>
> I'm a bit involved in the testing of the Inkscape 1.0 development branch, preliminary focussed on the Mac version. I just came across an [...]

where can i download dmg for inkscape 1.0alpha ?

eckmensch, 2019-08-30:
Hi Budi,

I'm taking this way to answer, don't have my github credentials handy in the office...

The link to download the dmg is in the issue description: https://inkscape.org/release/1.0alpha2/platforms/

Hope the mail will be delivered to you.

Best Regards

Patrick

badgeek, 2019-08-31:
hi @eckmensch, i  have problem when exporting each layer to png, here is the code
`
        for i in range(0, len(layer_arguments), EXPORT_PNG_MAX_PROCESSES):
            processes = []
            for layer_dest_svg_path, layer_dest_png_path, _, _, _ in layer_arguments[i:i+EXPORT_PNG_MAX_PROCESSES]:
                #export layer to png
                p = self.exportToPng(layer_dest_svg_path, layer_dest_png_path)
                processes.append(p)
            for p in processes:
                p.wait()
`

`

    def exportToPng(self, svg_path, output_path):
        area_param = '-D' if self.options.crop else 'C'
        command = "inkscape %s -d %s -e \"%s\" \"%s\"" % (area_param, self.options.dpi, output_path, svg_path)
        if (self.options.debug):
            inkex.debug(command)
        return subprocess.Popen(command.encode("utf-8"), shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)

`

inkscape doesnt seems to output any of the layer...

badgeek, 2019-08-31:
further debugging, it seems that export layer is not working

`
    def export_layers(self, dest, show):
        """
        Export selected layers of SVG to the file `dest`.
        :arg  str   dest:  path to export SVG file.
        :arg  list  hide:  layers to hide. each element is a string.
        :arg  list  show:  layers to show. each element is a string.
        """
        doc = copy.deepcopy(self.document)
        root = doc.getroot()
        for layer in doc.xpath('//svg:g[@inkscape:groupmode="layer"]', namespaces=inkex.NSS):
            id = layer.attrib["id"]
            if id in show:
                layer.attrib['style'] = 'display:inline'
            else:
                root.remove(layer)

        # remove the namedview for the hash as it changes based on user zoom/scroll
        namedview = doc.find('sodipodi:namedview', namespaces=inkex.NSS)
        root.remove(namedview)

        doc.write(dest)

        # returns a hash of the exported layer contents which can be used to
        # detect changes
        return hashlib.md5(ET.tostring(root)).hexdigest()`

badgeek, 2019-08-31:
testing using inkscape cli to export png:

`[:~ … tents/MacOS] [base] $ ./Inkscape-bin  -C -d 600.0 -e "/Users/xcorex/pcbINK4/.svg2shenzhen-cache/images/F.Cu_g40.png" "/Users/xcorex/pcbINK4/.svg2shenzhen-cache/csvaye8n.svg"

** (process:8073): WARNING **: 20:44:14.539: Units file /work/opt/share/inkscape/ui/units.xml is missing: Failed to open file ?/work/opt/share/inkscape/ui/units.xml?: No such file or directory

(process:8073): Gtk-WARNING **: 20:44:14.540: Locale not supported by C library.
        Using the fallback 'C' locale.

(<unknown>:8073): GdkPixbuf-WARNING **: 20:44:14.607: Cannot open pixbuf loader module file '/work/opt/lib/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
Try running the command
  gdk-pixbuf-query-loaders > /work/opt/lib/gdk-pixbuf-2.0/2.10.0/loaders.cache
to make things work again for the time being.
Cannot parse integer value ?600.0? for -d`

it seems there are bug from inkscape itself

eckmensch, 2019-08-31:
@badgeek Thanks for looking into that! I was able to replicate the described test on my system:

```shell
# Successful test of cli export with macports installation of Inkscape 0.92.4 (5da689c313, 2019-01-14) 
[mymac:~/Desktop] me% inkscape -C -d 600.0 -e export_cli_1.png 2.svg 
DPI: 600
Background RRGGBBAA: ffffff00
Area 0:0:900:900 exported to 5625 x 5625 pixels (600 dpi)
Bitmap saved as: export_cli_1.png

# Unsuccessful test of export with dmg based Inkscape 0alpha2 (863a58d, 2019-08-17), official notarized build
[mymac:~/Desktop] me% /Applications/Inkscape.app/Contents/MacOS/Inkscape-bin -C -d 600.0 -e export_cli_1.png 2.svg

** (process:6170): WARNING **: 23:26:08.620: Units file /work/opt/share/inkscape/ui/units.xml is missing: Failed to open file ?/work/opt/share/inkscape/ui/units.xml?: No such file or directory

(<unknown>:6170): GdkPixbuf-WARNING **: 23:26:08.754: Cannot open pixbuf loader module file '/work/opt/lib/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
Try running the command
  gdk-pixbuf-query-loaders > /work/opt/lib/gdk-pixbuf-2.0/2.10.0/loaders.cache
to make things work again for the time being.
Cannot parse integer value “600.0” for -d
```

I will reach out to the Mac developers in the Inkscape development team and come back once I know more...

dehesselle, 2019-09-02:
Hello! I'm leaving myself a note here that I have to look into this from Inkscape's perspective.

dehesselle, 2019-09-04:
I've got the export working in my test build. CLI interface has changed for 1.0, the correct command is
```
Inkscape -C -d 600 -o <output.png> <input.svg>
```
Note that 
- DPI has to specified as integer, not floating point
- output file has to specified with `-o`, not `-e`.
 
Use `Inkscape -?` to check for other relevant changes.

I'll probably get my fixes merged during the weekend and report back here once I have an official build ready.

badgeek, 2019-09-05:
@eckmensch @dehesselle  thanks!

eckmensch, 2019-09-05:
@dehesselle thanks for taking care and fixing that!

dehesselle, 2019-09-05:
Ready for testing.
```
Inkscape -C -d 600 -o <output.png> <input.svg>
```
http://alpha.inkscape.org/prereleases/Inkscape-1.0.0alpha2_190905_48e7c744.dmg

eckmensch, 2019-09-05:
@dehesselle This is brilliant! the version your provided works perfectly!

```bash
# Successful test of cli export with macports installation of Inkscape 0.92.4 (5da689c313, 2019-01-14) 

[mymac:~/Desktop] me% inkscape --version
Inkscape 0.92.4 (5da689c313, 2019-01-14)

[mymac:~/Desktop] me% inkscape -C -d 600.0 -e export_cli_1.png 2.svg 
DPI: 600
Background RRGGBBAA: ffffff00
Area 0:0:900:900 exported to 5625 x 5625 pixels (600 dpi)
Bitmap saved as: export_cli_1.png

# Successful test of cli export with dmg based Inkscape 1.0alpha2 (48e7c74, 2019-09-05)

[mymac:~/Desktop] me% /Applications/Inkscape.app/Contents/MacOS/Inkscape --version
Inkscape 1.0alpha2 (48e7c74, 2019-09-05)

[mymac:~/Desktop] me% /Applications/Inkscape.app/Contents/MacOS/Inkscape -C -d 600 -o export_cli_1.0alpha2.png 2.svg
Pango version: 1.42.3
Background RRGGBBAA: ffffff00
Area 0:0:900:900 exported to 5625 x 5625 pixels (600 dpi)
```

I have just one remark. The new Inkscape version seems to miss the final stdout statement:
```bash
Bitmap saved as: export_cli_1.png
```

badgeek, 2019-09-06:
thanks guys i will continue working on saturday cheers!

yiancar, 2020-01-05:
Hello!
Did we figure out a way to get vsg2shenzhen working on version 1 of inkscape?

ussjoin, 2020-01-06:
@badgeek Same question as above. I'm on 1.0beta2 (I'm on Catalina, so the previous version doesn't work on this computer anyway, and the SVG I'm working with crashes the stable (0.9.x) Inkscape when opened under a previous version of OSX) and it's not enormously clear how to modify what I'm doing to make svg2shenzen work. (Apologies if I'm missing something obvious, please feel free to point me to TFM if so.)

Prepare step throws an error but seems to do everything as expected; the export script has no output files. :-(

badgeek, 2020-01-06:
@ussjoin @yiancar unfortunately inkscape 1 support is pending due to changes in the inkscape api, if i have enough time i will work on it again

badgeek, 2020-01-06:
@ussjoin if export has no output files, maybe bitmap2component need to be chmod +x first

yiancar, 2020-01-06:
Thank for the reply @badgeek. As @ussjoin said, sadly, 0.9.x versions of inkscape no longer work on the newer versions of OSX.

badgeek, 2020-01-06:
@yiancar can you check the output of ./bitmap2component_osx in terminal ?

ussjoin, 2020-01-07:
[image]

@badgeek bitmap2component was already chmod +x, but when I ran it in the terminal, I got that popup from Gatekeeper; once I overrode it I got a more normal usage output.

```
./bitmap2component_osx

 bitmap2component-cli 

 Usage
 ./bitmap2component <filename.png> <output filename> <layer> <negative|true/false> <dpi> <threshold|0-255>
```

So with that fixed, I tried the export again. Still, sadly, no outputs of any kind from the export script. The below was the error it generated:

```export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:748: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:347: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 114, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 348, in processExportLayer
    layers = self.get_layers(curfile)
  File "export.py", line 548, in get_layers
    if  layer_label_name in self.layer_map.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'
```

badgeek, 2020-01-07:
@ussjoin its because you're on inkscape 1.0, for now you can use prev version of inkscape and try to run bmp2component again and enable it on gatekeeper

Mr-Blonde, 2020-01-25:
> AttributeError: 'dict' object has no attribute 'iterkeys'
> ```

The problem here is that export.py is written for Python2.7 and your Inkscape is invoking it with Python3 (there is no iterkeys() in Python3).

The solution to this is to explicitly tell Inscape to run this extension with Python2.7

Here is how:
https://stackoverflow.com/questions/52046845/changing-the-python-interpreter-for-inkscape

---

This fixes the above mentioned errors. On Mac OS 10.15 with Inkscape 1.0beta2 I am running into this error message now:

```
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 114, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 386, in processExportLayer
    prev_options = pickle.load(f)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1384, in load
    return Unpickler(file).load()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 864, in load
    dispatch[key](self)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 886, in load_eof
    raise EOFError
EOFError 
```

liquidcitymotors, 2020-04-14:
Hello!

Before seeing this, I went through the process of editing the export.py script to work with 1.0rc1 and python3. It didn't need too many changes, just the python version changes, the issue of the "output" argument in the pickled options, and the CLI syntax updates. 

I ran into the EOF error, that happens when the script exits before actually writing anything to options.pickle in the hidden cache folder. I had to either delete that file or handle an empty file in the script.

I ended up pickling a dict of the other options and then reading them back with prev_options as a dict instead of a namespace. That plus the tweaks to the CLI (-C instead of C, dpi as integer, -o instead of -e) and I was able to use this awesome tool with the native app on OSX 10.14. 

I could share my export.py if that helps (I didn't yet update deprecated functions but I could...). Thanks for making this @badgeek !

liquidcitymotors, 2020-04-15:
great! this is my first contribution to an open source project so I wasn't sure how to proceed. you can find the changes I tried here: 

https://github.com/liquidcitymotors/svg2shenzhen/commit/f884a15ee64ada80548fb645baa9e89f2f357ee7

I am not confident enough with them to do a pull request, but this is working on macos 10.14.5 with inkscape 1.0rc1 and the latest nightly of kicad. let me know if I can do anything else to help here 😃

badgeek, 2020-08-17:
@liquidcitymotors inkscape 1.0 will be supported on v.18 i will release it soon thank you

closed 2020-08-17 by badgeek