badgeek/svg2shenzhen #51: Doesn't work under 0.92.4 on Lubuntu 19.03

Juul, 2019-10-24:
Fails during export with a `No such file or directory: <export_path>/.svg2shenzhen-cache/drawing.pretty/F.Cu_g19.kicad_mod`

Where `<export_path>` is the path the user specified as the export location.

This is svg2shenzhen version 0.2.17 from the pre-compiled zip file.

Someone else was having a different issue with this version of kicad on macOS in issue #44. Unfortunately 0.92.4 is the version of Kicad in the Lubuntu/Ubuntu 0.92.4 apt repository.

Juul, 2019-10-24:
I just tested this using the latest code from github compiled from source and it gives the same error.

The error is in `processExportLayer` on line 490 in commit e64598ad078543a487c703024ffdf17311aa92aa

Juul, 2019-10-24:
Never mind. I must have had some issue with my installation. Re-installed everything and it all works.

ccoenen, 2019-11-14:
I have the same problem, do you remember anything specific that might need to be changed @Juul ?

Juul, 2019-11-15:
@ccoenen Are you using the latest version compiled from source from github? Be aware that the repo includes pre-compiled binaries and when you compile the resulting binary ends up in a different spot. I installed the stable release version, then compiled the binary from the latest code on git, then copied the resulting compiled binary over the previous linux binary in the inkscape plugin dir, then restarted inkscape.

ccoenen, 2019-11-15:
The one linked from "releases", not the zip from master. But I'll try compiling at a different time. I also tried svg2mod, and that worked mostly "OK" for me, so right now I'm not in a great hurry.

badgeek, 2019-11-15:
@Juul @ccoenen i think i need to upload tar.gz version since it preserve chmod +x on the bitmap2component executable... the zip on release is already correct i think

closed 2020-08-17 by badgeek