badgeek/svg2shenzhen #58: Error when attempting to export to Kicad

cglmrfreeman, 2019-12-02:
The error occurs when clicking "Apply" on the "2. Export Kicad" window when exporting as a "KICAD - Module":
>`Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "C:\Program Files\Inkscape\share\extensions/inkex.py", line 283, in affect
    self.effect()
  File "export.py", line 322, in effect
    self.setInkscapeScaling()
  File "export.py", line 286, in setInkscapeScaling
    viewbox = root.attrib['viewBox'].split(' ')
  File "src/lxml/etree.pyx", line 2455, in lxml.etree._Attrib.__getitem__
KeyError: 'viewBox'`

I am unsure how to proceed, please advise.

badgeek, 2020-08-17:
what is your os, inkscape version?

cglmrfreeman, 2020-08-17:
OS was Windows 10 x64 Home (however this is no longer my daily driver) and Inkscape version 0.92.4 (thankfully not updated since this bug was filed).

badgeek, 2020-08-17:
please update your inkscape to v1 and i just released svg2sz v0.2.18 that only work inkscape v1

cglmrfreeman, 2020-08-18:
Updated. This issue no longer exists. I will create new issues as I find them. Thank you for your continued support in this project.

closed 2020-08-18 by cglmrfreeman