badgeek/svg2shenzhen #61: Only 'help' available from extension

ehafh, 2020-01-10:
Installed inkscape and svg2shenzhen. When navigating to the extension in inkscape it only has option 0 'help'. Prepare and export do not show up at all. Have tried with a new inkscape blank file, and example svg files plus my own. 

Any idea's?
Win 10

Mr-Blonde, 2020-01-24:
I am having the same exact problem (with no solution).

I am on Inkscape 1.0beta2 (only available version for Mac OS 10.15). Does the plugin not work with version 1.0?

Mr-Blonde, 2020-01-25:
I guess, I found the problem with my installation. The extension log file was complaining that it couldn't find the Python scripts. 

_What I did to resolve it:_
I copied the scripts (+ any other files) from the svg2shenzhen subdirectorie into the extension's main directory.

i.e.:
**from <path to Inkscape extensions>/svg2shenzhen-extension-0.2.7/svg2shenzhen
to <path to Inkscape extensions>/svg2shenzhen-extension-0.2.7/**

Restarting Inkscape resulted in the two menu items being not grayed out anymore.

Now, however, I have the issue that the scripts are Python 2.x and Inkscape invokes them with Python 3. While preparing the PCB still works (with depreciation warnings), the export fails with multiple error messages (which I wasn't able to fix so far).

badgeek, 2020-08-17:
@Mr-Blonde for inkscape v.1 check out latest svg2sz update v.18

closed 2020-08-17 by badgeek