badgeek/svg2shenzhen #63: Different solder mask clearance?

Moonbase59, 2020-02-08:
I need to produce a "shine-through" PCB (text with LEDs behind). I can put an exact duplicate of my text on the F.Cu and F.Mask layers but the letters on the Mask will always be "bigger" (i.e. some solder mask clearance is auto-added). Changing the solder mask clearance in PCBnew afterwards doesn’t help.

Is there any trick or workaround to get a zero solder mask clearance?

arduino12, 2020-04-03:
Hi,

I think this is our problem (taken from the generated kicad file):
[image]

**`svg2shenzhen` needs to set the `solder mask clearance` to 0 before adding the layers...**

How can we fix it in the code? (setting it after the script doesn't work..)

Thanks,
Arad

rcludwick, 2020-05-02:
Also having played around with that, it's looks like on kicad 5.1.5 the solder mask isn't rendered correctly until you generate the gerbers.  So the 3d viewer doesn't respect it apparently.

closed 2020-08-17 by badgeek