badgeek/svg2shenzhen #64: Error on export

besi, 2020-02-09:
```
export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:748: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:347: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 114, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 397, in processExportLayer
    pickle.dump(options, f)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1376, in dump
    Pickler(file, protocol).dump(obj)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 224, in dump
    self.save(obj)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 331, in save
    self.save_reduce(obj=obj, *rv)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 425, in save_reduce
    save(state)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 286, in save
    f(self, obj) # Call unbound method with explicit self
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 655, in save_dict
    self._batch_setitems(obj.iteritems())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 669, in _batch_setitems
    save(v)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 306, in save
    rv = reduce(self.proto)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy_reg.py", line 70, in _reduce_ex
    raise TypeError, "can't pickle %s objects" % base.__name__
TypeError: can't pickle file objects

```

yene, 2020-02-09:
same problem here macos 10.15.2

besi, 2020-02-09:
There is a similar pickle related issue #49 but it's not the exact same error message.

kasbah, 2020-02-09:
A quick work around, to get ahead in the workshop could be to disable caching by commenting out the two lines that try and pickle the options:

https://github.com/badgeek/svg2shenzhen/blob/e64598ad078543a487c703024ffdf17311aa92aa/inkscape/svg2shenzhen/export.py#L396-L397

badgeek, 2020-08-17:
@besi @yene this will be fixed in v.18 i will soon release it

closed 2020-08-17 by badgeek