badgeek/svg2shenzhen #72: Edge cuts never align

mplacona, 2020-05-04:
Hey, I followed @twinkletwinkie's videos and everything has worked beautifully other than the fact that when I try to have multiple layers under a layer and try to export I get an error with the plugin. I went around this by just having everything on the same layer (i.e. registration and board outline all under Edge.Cuts) and it's all good.

My problem though is that I am unable to get the Edge cuts aligned with the rest of the board items (solder mask, copper etc).

Everything else gets imported aligned, but not the edge cuts. I am wondering if I am doing anything wrong on my layers or if this is to be expected.

My Front Silk runs along the edges of the board, and right now I get a tiny gap sometimes due to the misalignment.

Any suggestions here much appreciated.

kasbah, 2020-05-04:
Just a quick guess, not sure at all: maybe you have put things in a group on the edge cut layer? Groups need to be avoided on edge cuts and drill layers.

mplacona, 2020-05-04:
> Just a quick guess, not sure at all: maybe you have put things in a group on the edge cut layer? Groups need to be avoided on edge cuts and drill layers.

Pretty sure I will have done that. What's the alternative in Inkspace?

kasbah, 2020-05-04:
It's not really about alternatives, currently you need to ungroup everything on the drill and edge cut layers before exporting using svg2shenzhen (unless this has been improved since I last looked at the code).

mplacona, 2020-05-04:
That totally did it @kasbah! Thank you so much! I ungrouped and they all came aligned together!

closed 2020-05-04 by mplacona