badgeek/svg2shenzhen #74: not working in Inkscape 1.0

okelk, 2020-05-14:
export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:748: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:347: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\deprecated.py", line 180, in affect
    return self.run(args=args)
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 348, in processExportLayer
    layers = self.get_layers(curfile)
  File "export.py", line 548, in get_layers
    if  layer_label_name in self.layer_map.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'

giuliomoro, 2020-05-14:
confirmed

hhaudio, 2020-05-15:
confirmed

hhaudio, 2020-05-15:
I am getting this error as well on Inkscape 1.0 OSX 10.14.6

hhaudio, 2020-05-15:
export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:748: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:347: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/deprecated.py", line 180, in affect
    return self.run(args=args)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 348, in processExportLayer
    layers = self.get_layers(curfile)
  File "export.py", line 548, in get_layers
    if  layer_label_name in self.layer_map.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'

fstokke, 2020-05-17:
Inkscape made some breaking changes in the extension API in version 1.0. See https://wiki.inkscape.org/wiki/index.php/Updating_your_Extension_for_1.0

I have tried to upgrade the extension myself. I have fixed optparse warnings and some other problems, but I'm now stuck at this one that happens during export:

export.py:749: DeprecationWarning: inkex.bezier.cspsubdiv -> Sub-divide cubic sub-paths
  cspsubdiv.cspsubdiv(p, 0.01)
export.py:759: DeprecationWarning: simplepath.formatPath -> str(element.path) or str(Path(array))
  node.set('d', simplepath.formatPath(np))
export.py:653: DeprecationWarning: simplepath.parsePath -> element.path.to_arrays()
  p = simplepath.parsePath(d)
export.py:671: DeprecationWarning: simpletransform.applyTransformToPoint -> Transform(mat).apply_to_point(pt)
  simpletransform.applyTransformToPoint(trans,xy)
export.py:734: DeprecationWarning: simpletransform.applyTransformToPoint -> Transform(mat).apply_to_point(pt)
  simpletransform.applyTransformToPoint(trans,pt)

fstokke, 2020-05-19:
Inkscape has changed the extension API in version 1.0. I have been working on a patch to svg2shenzhen to adapt to the new API. I think I got it working today :) I will clean up the code and submit a pull request.

daskycodes, 2020-05-19:
I was able to make it work on MacOS Catalina 10.15.3 with the help of the following guide and some additional changes

[Updating your Extension for 1.0](https://wiki.inkscape.org/wiki/index.php/Updating_your_Extension_for_1.0)

https://github.com/daskycodes/svg2shenzhen/commit/b0dbf671f4559c7dd3245821dc44d9617c5ee95c

The export works but still gives me the following messages

```
export.py:323: ResourceWarning: unclosed file <_io.BufferedReader name=5>
  self.processExportLayer()
ResourceWarning: Enable tracemalloc to get the object allocation traceback
export.py:323: ResourceWarning: unclosed file <_io.BufferedReader name=7>
  self.processExportLayer()
ResourceWarning: Enable tracemalloc to get the object allocation traceback
```

fstokke, 2020-05-21:
I have submitted a pull request with support for Inkscape 1.0 (#76). Hopefully someone on the development team can review it.

badgeek, 2020-08-17:
this will be fixed in v .18 and release is really soon

closed 2020-08-17 by badgeek