badgeek/svg2shenzhen #75: bitmap2component puts F.SilkS layer onto F.Cu layer in generated footprint

fstokke, 2020-05-21:
I have been running svg2shenzhen from the master branch (065be79). It seems like there is a problem with the precompiled bitmap2component in the inkscape folder. It puts the F.SillkS layer onto the F.Cu layer in the generated kicad_mod footprints.

I'm running Inkscape 0.92.5 on Linux. I have also tested on Windows and same problem there. The bitmap2component.exe from the 0.2.17 release seemed to work OK on Windows.

I have attached a .zip with the SVG and generated footprint.

[svg2shenzhen-issue-report.zip](https://github.com/badgeek/svg2shenzhen/files/4661575/svg2shenzhen-issue-report.zip)

badgeek, 2020-05-28:
hi @fstokke is this already in your PR ?

fstokke, 2020-05-28:
Hi @badgeek. No it's not in my pull request. In the pull request I just reverted the .exe for windows and replaced it with the one you released in the zip for the 0.2.17 release.

But the problem also applies to the pre-compiled Linux and Mac executables that are currently in the master branch.

I don't have the development setup for building the executables myself. Is this something you can fix? I can update my pull request once the executables in master branch have been fixed (i.e. remove the updated Windows .exe from the pull request)

fstokke, 2020-05-28:
I see that you have merged my pull request. Thanks :)

The executables for all plattforms probably need to be updated with correct handling of the silk screen layer.

amcewen, 2020-08-03:
I've been running the version from https://github.com/badgeek/svg2shenzhen/commit/77eda90fd8ec4fa33a9d1ba870d9fee403698fcb so I could use it with Inkscape 1.0 (thanks for the PR @fstokke!) and was hitting this problem with the pre-built `inkscape/svg2shenzhen/bitmap2component_linux64`.

I re-built it from source and that seems to be working okay - I've not done lots of testing on it, but it is putting things onto the silkscreen rather than the copper :-)

badgeek, 2020-08-17:
@amcewen check v.18 i will release it soon, this bug already fixed

closed 2020-08-17 by badgeek