badgeek/svg2shenzhen #77: Unable to export Inkscape 1.0 MacOS

Xcreed, 2020-05-27:
I've been following the steps from #74 but now I'm getting this error: 

```Traceback (most recent call last):
  File "export.py", line 749, in <module>
    _main()
  File "export.py", line 745, in _main
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 321, in effect
    self.processExportLayer()
  File "export.py", line 423, in processExportLayer
    hash_sum = self.export_layers(layer_dest_svg_path, show_layer_ids)
  File "export.py", line 514, in export_layers
    root.remove(layer)
  File "src/lxml/etree.pyx", line 944, in lxml.etree._Element.remove
ValueError: Element is not a child of this node.

Not sure if this is caused by the relative path. I'm using ~/Documents, but have also tried the absolute path.

fstokke, 2020-05-27:
Have you tried the updated [export.py](https://github.com/badgeek/svg2shenzhen/blob/f34105e6eecb56c29af639a1eef4df94befa39d3/inkscape/svg2shenzhen/export.py) I submitted pull request #76?

Xcreed, 2020-05-27:
I thought I did. I just used your code for export.py and now I'm getting this error. Tried Kicad PCB and Module, same results. 

```
Traceback (most recent call last):
  File "export.py", line 741, in <module>
    _main()
  File "export.py", line 737, in _main
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 319, in effect
    self.processExportLayer()
  File "export.py", line 459, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 650, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range
```

fstokke, 2020-05-27:
Are you able to reproduce the error with any of the example svgs shipped with svg2shenzhen?
Does it work with the previous version of Inkscape/svg2shenzhen?

Xcreed, 2020-05-28:
Going back to this. With the virus.svg example from the repo it works. It makes me believe it's an error with one of my layers since using a blank document works just fine. Trying to debug this, any idea how to get this to print so I can check where the problem is? 

```for layer in doc.xpath('//svg:g[@inkscape:groupmode="layer"]', namespaces=inkex.NSS):
            id = layer.attrib["id"]
            if id in show:
                layer.attrib['style'] = 'display:inline'
            else:
                print("layer to remote" + str(layer.attrib["id"]))
                root.remove(layer)
```

Xcreed, 2020-05-28:
Going back to this. The virus.svg file works which makes me feel the error is on my file. I keep getting the same error:

``` File "export.py", line 749, in <module>
    _main()
  File "export.py", line 745, in _main
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 321, in effect
    self.processExportLayer()
  File "export.py", line 423, in processExportLayer
    hash_sum = self.export_layers(layer_dest_svg_path, show_layer_ids)
  File "export.py", line 514, in export_layers
    root.remove(layer)
  File "src/lxml/etree.pyx", line 944, in lxml.etree._Element.remove
ValueError: Element is not a child of this node.
```
Turns out this error isn't thrown if there's nothing nested inside the layers. My layout below was triggering the error.

[image]

After removing the nested files it kinda worked. For some reason having path objects in the Edge.Cuts layer returns the index out of range. 

[image]

Having a regular object (filled or just the stroke) works just fine. 

[image]

Those are my findings at the moment. No idea how to turn a path into an object, so pretty much need to import the Edge.Cuts object otherwise won't export. I can confirm I get the same error `lxml.etree._Element.remove
ValueError: Element is not a child of this node.` when using an older version either on Linux or Windows.

Xcreed, 2020-05-28:
Update: No idea what's going on with the Edge.Cuts issue, tried again and worked just fine. One thing I realized, when taking things out of the "nested" groups, you need to the remove that group object otherwise it will continue to give errors.

fstokke, 2020-05-29:
Great that you got it working :) I have not had any problems with exporting paths in the Edge.Cuts layer. I have also noticed the problem with the silk screen being put on the Cu layer, see Issue #75.

badgeek, 2020-08-17:
@Xcreed fcu and fsilk bug will be fixed in v.18, realease is real soon

closed 2020-08-17 by badgeek

LArkema, 2022-05-21:
For anyone else who comes along and gets the same error, I also found out after a long and painful trial-and-error process that svg2shenzhen doesn't allow for sub-layers under its designated layers - so any cluster of objects needs to be a _group_, not a layer. In the Objects menu, this looks like the condensed stack of paper with a golden paper on top (group) instead of a loose stack of three white papers (layer) under the 'T' column. I had never used Inkscape before, so it took me a while to figure out what may have been an obvious error. 

In the screenshot below, F.mask has to be a layer, but g8309 and Train Lines have to be groups.
[image]