badgeek/svg2shenzhen #78: Error on export on KiCAD

hyuma78, 2020-06-18:
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 289, in affect
    self.effect()
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 474, in processExportLayer
    f.write(kicad_drill_string)
TypeError: expected a string or other character buffer object

hyuma78, 2020-06-18:
also:
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 289, in affect
    self.effect()
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 467, in processExportLayer
    with open(kicad_file, 'r') as f:
IOError: [Errno 2] No such file or directory: '/home/hyuma/iron giant.jpg.2020_06_18_21_48_44.0.pretty/F.Cu_g390.kicad_mod'

hyuma78, 2020-06-18:
I tried debug mode with other image:

inkscape C -d 600.0 -e "/home/hyuma/.svg2shenzhen-cache/images/F.Cu_g84.png" "/home/hyuma/.svg2shenzhen-cache/sCZ5tD"
"/usr/share/inkscape/extensions/svg2shenzhen/bitmap2component_linux64" "/home/hyuma/.svg2shenzhen-cache/images/F.Cu_g84.png" "/home/hyuma/ironman-face-png-7.pretty/F.Cu_g84.kicad_mod" F.Cu true 600 5
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 289, in affect
    self.effect()
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 467, in processExportLayer
    with open(kicad_file, 'r') as f:
IOError: [Errno 2] No such file or directory: '/home/hyuma/ironman-face-png-7.pretty/F.Cu_g84.kicad_mod'

hyuma78, 2020-06-18:
I made some chmod + on files inside /usr/share/inkscape/extensions

and now:

inkscape C -d 600.0 -e "/home/hyuma/.svg2shenzhen-cache/images/F.Cu_g84.png" "/home/hyuma/.svg2shenzhen-cache/5zbA87"
"/usr/share/inkscape/extensions/svg2shenzhen/bitmap2component_linux64" "/home/hyuma/.svg2shenzhen-cache/images/F.Cu_g84.png" "/home/hyuma/ironman-face-png-7.pretty/F.Cu_g84.kicad_mod" F.Cu true 600 5

badgeek, 2020-08-17:
what is your inkscape version?