badgeek/svg2shenzhen #80: silkscreen is exported as copper

x70b1, 2020-07-24:
I found out that `svg2shenzhen` translates the silkscreen layer to a copper layer. But thats obviously wrong.
To fix this I can manually edit the layer in the exported `.kicad_pcb` file. That would work.

Maybe it was my mistake. I tried a lot of things. But I always came to the same conclusion.

The `svg2shenzhen` was build from the latest commit on master branch. 
Inkscape is 1.0 (4035a4fb49, 2020-05-01).

The example:
[image]
[image]

jollydodo, 2020-07-27:
Same problem here. Any idea what could be causing this issue? (Using MacOS 10.15.6 and the latest commit)

fstokke, 2020-07-31:
I think this is a problem with the latest builds of bitmap2component.exe, see issue #75. On Windows I tried reverting to the previous version and that one worked OK.

x70b1, 2020-08-01:
@fstokke do you know what is required to fix this?

fstokke, 2020-08-03:
I see @amcewen has fixed the problem on Linux by recompiling bitmap2component from source (see issue #75), so maybe a recompile will fix the problem on Mac as well. Unfortunately I don't have the build environment for compiling from source.

badgeek, 2020-08-17:
@x70b1 @fstokke this will be fixed in v.18 with your PR @fstokke thank you, with new build system using travis i hope its easier to update

closed 2020-08-17 by badgeek