badgeek/svg2shenzhen #85: Drill sizes too big after exporting to KiCad

atomworks, 2020-08-18:
When I export my drilling layout in correct but the drilling sizes are all 10x too big. For example, I have holes that are 6mm in Inkscape but are 60mm once in KiCad. The positioning is correct however... is it something I'm doing wrong?

I'm on Inkscape v1.0 with the very latest 0.2.18.3 release of svg2shenzhen though I was having the same issue before I updated those today.

twinkletwinkie, 2020-08-18:
Can you share a svg with just the drills please?

atomworks, 2020-08-19:
Sure thing, here you go.

[drill.svg.zip](https://github.com/badgeek/svg2shenzhen/files/5093865/drill.svg.zip)

badgeek, 2020-08-20:
[image]

i copy paste your drill to another svg file (viruspcb) and its correctly sized, can you prepare a new file then export again?

twinkletwinkie, 2020-08-20:
I'm seeing a lot of holes called "3.5mm" but they're showing 6x6mm.

[image]

atomworks, 2020-08-20:
@badgeek I've tried copying it to new documents a few times. Each time though I export the SVG with the drill data that has actually been generated by a different artwork package. I'll try to copy it from Inkscape to Inkscape if that seems to cause you no issues.

@twinkletwinkie Sorry for the confusion there. It says 3.5mm because the holes are for 3.5mm audio jacks (which are 6mm wide).

twinkletwinkie, 2020-08-20:
> @badgeek I've tried copying it to new documents a few times. Each time though I export the SVG with the drill data that has actually been generated by a different artwork package. I'll try to copy [...]

@atomworks I found your problem.  I think your program you're creating it in is adding more data to the circles than intended.  SVG files are plain text so you can review their source and all of the circles in that document has
`transform="scale(0.1)"` on them but when I drew another circle it does not have that transform property.   If I do what @badgeek did and copy/paste the circles into another Inkscape document and save that the `transform="scale(0.1)"`  property does not come over with it, whether or not it's saved as an Inkscape SVG or a Plain SVG.

atomworks, 2020-08-21:
Thanks for the information @twinkletwinkie. I took a look in to the files I'm importing in to Inkscape and their really minimal however because my source program works in px, when I import the SVG and resize it to mm Inkscape adding the scaling as part of this. I'm sure the previous version on Inkscape handled it differently perhaps as I used this workflow for a start and didn't have this issue. 

Thinking about it though, it might also be because it's essentially a group when I resize it and then once sized I ungroup it. Perhaps in that instance it's doing `transform="scale(0.1)"` on the group and then commuting that to the child layers when it's ungrouped.

I think I need to work out a way to get my SVGs to scale properly on import without the manual work. That way I can bypass those problems whilst generally maintaining a better practice.

Thanks for your assistance on this. Perhaps the `transform="scale(0.1)"` thing is something you want to consider in the future but I expect likely not as it comes from outside of the intended workflow. Either way, I'll close this issue as the solution is on my end.

closed 2020-08-21 by atomworks

atomworks, 2020-08-24:
Some information for anyone else who might arrive here with a similar problem...

It seems that Inkscape is generating the `transform="scale(0.1)"` code which is causing the problem. It seems that this is part of the way it imports SVGs. This isn't a problem if you use Inkscape for everything in the first place but that's not how I prefer to do it.

So, if you have this issue, once you've saved your Inkscape SVG you can go in to the SVG code in a code or text editor, do a search for the drill layer and then manually delete the `transform="scale(0.1)"` and then add a decimal to the size of the layers nested within the drill layer to re-correct the size and spacing.