badgeek/svg2shenzhen #88: Export Error with multiple art in F.SilkS

iharshadec, 2020-08-19:
svg2shenzhen - 0.2.18.5
inkscape - 1.0

Export Settings
[image]

Export Error
[image]

.svg file
[pcbArt.zip](https://github.com/badgeek/svg2shenzhen/files/5097665/pcbArt.zip)

badgeek, 2020-08-20:
@iharshadec i just checked your file and the layers is not properly named, i need file with proper layer

iharshadec, 2020-08-20:
I have generated layers with prepare document options only.
There is all layers that I did checkboxes while preparing document. In addition to that I have added Extra layer to put some extra/tirals on it.

Layers
[image]

Objects
[image]

iharshadec, 2020-08-20:
I got this working. I think there was some problem with my antivirus which was blocking bitmap2component.exe file and because of that this error was coming there.

closed 2020-08-20 by iharshadec