badgeek/svg2shenzhen #89: Extension menu items (prepare document & export to kicad) are disabled

herpiko, 2020-08-19:
I'm using the latest version (0.2.18.5). However the About menu is not disabled and can be clicked, showing the about dialog as expected. I've tried to downgrade the version but it didn't work either.

### Ubuntu 18.04 with Inkscape 1.0 (1.0+r73+1) from PPA (ppa:inkscape.dev/stable)
[image]

### macOS Mojave with Inksape 1.1-dev (48d18c81, 2020-07-08)
[image]

badgeek, 2020-08-20:
@herpiko Hi, we havent test the extension with inkscape 1.1-dev yet, please try with inkscape 1.0 instead, why would you use dev version of inkscape?

badgeek, 2020-08-20:
please use svg2sz latest v0.2.18.6 https://github.com/badgeek/svg2shenzhen/releases/tag/0.2.18.6

herpiko, 2020-08-20:
@badgeek Thanks for your fast response.

Ok, I'll try it with 0.2.18.6.

aa956, 2020-08-21:
0.2.18.6 just behaved the same way on Windows when installed into svg2shenzhen subdirectory of Inkscape user extensions directory:

So menu items disabled if extension unpacked to:
%USERPROFILE%/inkscape/extensions/svg2shenzhen/

menu items enabled if extension unpacked directly to:
%USERPROFILE%/inkscape/extensions/

badgeek, 2020-08-21:
@aa956 what is your inkscape version?

aa956, 2020-08-21:
Inkscape version: 1.0 (4035a4fb49, 2020-05-01)

badgeek, 2020-08-21:
directory layout should be like this :
[image]

i think i need a better install instructions

aa956, 2020-08-21:
Another option is to change inx file from:
```
    <script>
        <command reldir="extensions" interpreter="python">svg2shenzhen/export.py</command>
    </script>
```
to
```
    <script>
        <command reldir="inx" interpreter="python">svg2shenzhen/export.py</command>
    </script>
```
as recommended here: [Updating your Extension for 1.0
](https://wiki.inkscape.org/wiki/index.php/Updating_your_Extension_for_1.0)

If I understand correctly this change will enable installation anywhere under extensions directory?