badgeek/svg2shenzhen #90: Prepare document throw error

badgeek, 2020-08-20:
[image]

badgeek, 2020-08-20:
@twinkletwinkie  kicadLayers.items returns dict_keys which need to be wrapped with list(... )

badgeek, 2020-08-20:
https://github.com/badgeek/svg2shenzhen/blob/0dc743cd4518bfae795154d72c53b5ebeff8f7e2/inkscape/svg2shenzhen/prepare.py#L172

badgeek, 2020-08-20:
- Fixed in https://github.com/badgeek/svg2shenzhen/releases/tag/0.2.18.6

closed 2020-08-20 by badgeek

twinkletwinkie, 2020-08-20:
Strange. I don’t understand why that would throw an error now when it doesn’t on my computers.

badgeek, 2020-08-20:
what is your OS/inkscape version? @twinkletwinkie

badgeek, 2020-08-20:
this is the tricky part with python inside inkscape, that is why i have big mess of ifs to keep the code minimal, i hope this will not break again

twinkletwinkie, 2020-08-20:
Windows 10 1909 and Inkscape 1.0

badgeek, 2020-08-20:
For note: broken on Inkscape 1.0 (4035a4f, 2020-05-01) OSX 10.13.6 (17G11023) High Sierra