badgeek/svg2shenzhen #100: Export error

GuillermoTell, 2020-08-26:
Hi, no idea about what's going on here. It doesn't finish exporting the module and gives this error:

Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 501, in processExportLayer
    kicad_modules_string += kicad_drill_string
TypeError: can only concatenate str (not "NoneType") to str

The virus example file works well.

GuillermoTell, 2020-08-26:
I don't know why, but if you enable all the layers when you prepare the document it works. I was using just Edge.Cuts, F.Cu and F.Mask-auto.

badgeek, 2020-08-26:
for now drill layer is mandatory, i will fix on weekend

badgeek, 2020-08-30:
this is fixed in latest v 0.8.1.7 @GuillermoTell

closed 2020-08-30 by badgeek