badgeek/svg2shenzhen #102: kicad_drill_string error

waltsatan, 2020-08-27:
I'm using the latest svg2shenzhen with Inkscape 1.0 on Mac OS 10.12 and am getting this error when I try to export a F.SilkS layer:

```
Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 501, in processExportLayer
    kicad_modules_string += kicad_drill_string
TypeError: can only concatenate str (not "NoneType") to str
```
I've tried multiple variations of different layers and also tried exporting a simple basic square shape, but get the same error.  I added a blank Drill layer after noticing "kicad_drill_string", and there's no error, but the .kicad_mod file is empty except for a couple of lines default fp_text references (Ref** and Val**).

Anyone encounter this error?

badgeek, 2020-08-28:
please check the drill layer, it is mandatory until i fix on next release

badgeek, 2020-08-30:
this is fixed in latest v 0.8.1.7 @waltsatan

closed 2020-08-30 by badgeek

waltsatan, 2020-09-02:
Confirming the fix. Thanks!