badgeek/svg2shenzhen #103: Edge.Cuts flatten Bezier issue

GuillermoTell, 2020-09-05:
Hi, I don't know why, but I get this message:

Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 468, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 659, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range

I'm using a simple rectangle as Edge.Cuts, already transformed into path. However, if I check the flatten Bezier option every time, it works.

badgeek, 2020-09-06:
do you have example svg file ?

GuillermoTell, 2020-09-07:
Here it is:
[Test.zip](https://github.com/badgeek/svg2shenzhen/files/5182608/Test.zip)

iharshadec, 2020-09-08:
Here is your updated Test.zip. I am able to export properly.
1. I am not sure if you should do F.Mask-inverted you can leave F.Mask as it is.
2. You must do Prepare document first to make sure your drawing is within that area only.
[Test.zip](https://github.com/badgeek/svg2shenzhen/files/5185930/Test.zip)