badgeek/svg2shenzhen #104: No output file-Inkscape message error

jcarolinares, 2020-09-07:
Hello, I'm triying to use the extension but I'm having problems with the "Export to kicad" step. When I prepare the test viruspcb and I click on export as a module and I click on apply Inskcape return the following Python traceback from the extension:

`Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/snap/inkscape/7921/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 497, in processExportLayer
    with open(kicad_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jcarolinares/snap/inkscape/7921/.svg2shenzhen-cache/viruspcb.pretty/F.Cu_g8165.kicad_mod'`

My setup is:

- Ubuntu 16.04
- Snap packet of Inkscape. Inkscape version 1.0

Any idea about how to fix this problem? Thank you!

iharshadec, 2020-09-08:
@jcarolinares can you share your file?

jcarolinares, 2020-09-11:
Hello! 

It happens with every .svg file, but in this case, the log is from the example file "viruspcb" found on the readme of this repo:

[https://raw.githubusercontent.com/badgeek/svg2shenzhen-next/master/examples/viruspcb.svg](https://raw.githubusercontent.com/badgeek/svg2shenzhen-next/master/examples/viruspcb.svg)

I've tried to figure myself what is happening but no luck.

Regards!

iharshadec, 2020-09-11:
@jcarolinares 
I just exported without any error in windows. By looking at your error looks like you have **Open KiCad** option checked.
Try one more time by unchecking all options under Export to Kicad.

Something Like this....
[image]

jcarolinares, 2020-09-11:
Hi.

Sorry, but I'm using Ubuntu 16.04 as I commented above... Yes, of course, it works on windows, the problem is on Ubuntu 16.04 :+1: 

The error happens with the "open on Kicad", flatten bezier, and PCBway unchecked.

Regards

badgeek, 2020-09-11:
can you screenshot me "Export to Kicad" dialog ?

rcludwick, 2020-11-22:
It looks like there's an error preventing the file in the cache from being created. 

I saw this when bitmap2component_linux64 wasn't found...