badgeek/svg2shenzhen #108: Error export []

Noisk8, 2020-10-14:
Hi all, im try export a proyect bun have this error

Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/snap/inkscape/7947/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 497, in processExportLayer
    with open(kicad_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/ink_ext_XXXXXX.pretty/B.Cu_g20.kicad_mod'

i m use pop os | inkscape 1.0.1 and de the las version of the svg2shenzhen

somewhere have a idea?

iharshadec, 2020-10-14:
Can you share your svg file?

hodgesds, 2020-10-25:
I was able to reproduce with the svg in the examples directory:

```
inkscape -C -d 600 -o "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Cu_g3354.png" "/home/daniel/pcb/wut/.svg2shenzhen-cache/tcz6y47n.svg"
inkscape -C -d 600 -o "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Mask_g3356.png" "/home/daniel/pcb/wut/.svg2shenzhen-cache/wo4pzglk.svg"
inkscape -C -d 600 -o "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Silk_g3358.png" "/home/daniel/pcb/wut/.svg2shenzhen-cache/_zhn3v92.svg"
"/home/daniel/.config/inkscape/extensions/svg2shenzhen/bitmap2component_linux64" "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Cu_g3354.png" "/home/daniel/pcb/wut/myFirstPCB.pretty/F.Cu_g3354.kicad_mod" F.Cu true 600 128
"/home/daniel/.config/inkscape/extensions/svg2shenzhen/bitmap2component_linux64" "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Mask_g3356.png" "/home/daniel/pcb/wut/myFirstPCB.pretty/F.Mask_g3356.kicad_mod" F.Mask true 600 128
"/home/daniel/.config/inkscape/extensions/svg2shenzhen/bitmap2component_linux64" "/home/daniel/pcb/wut/.svg2shenzhen-cache/images/F.Silk_g3358.png" "/home/daniel/pcb/wut/myFirstPCB.pretty/F.Silk_g3358.kicad_mod" F.SilkS true 600 128
Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/usr/share/inkscape/extensions/inkex/base.py", line 123, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 474, in processExportLayer
    with open(kicad_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/daniel/pcb/wut/myFirstPCB.pretty/F.Cu_g3354.kicad_mod'
```