badgeek/svg2shenzhen #111: Broken with new Inkscape API changes

f0x52, 2020-11-20:
'Prepare Document' gives the following error on Inkscape 1.0.1 (3bc2e813f5, 2020-09-07)
```
prepare.py:24: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--docwidth", action="store", type="float", dest="docwidth", default=0.0)
prepare.py:25: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--docheight", action="store", type="float", dest="docheight", default=0.0)
prepare.py:334: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
prepare.py:65: DeprecationWarning: inkex.etree was removed, use "from lxml import etree"
  layer = inkex.etree.SubElement(svg, 'g')
prepare.py:71: DeprecationWarning: inkex.etree was removed, use "from lxml import etree"
  rect = inkex.etree.Element(inkex.addNS('rect','svg'))
prepare.py:77: DeprecationWarning: simplestyle.formatStyle -> str(inkex.Style(a))
  rect.set('style', formatStyle(style))
prepare.py:203: DeprecationWarning: inkex.etree was removed, use "from lxml import etree"
  grid = inkex.etree.Element(inkex.addNS('grid','inkscape'))
```

'Export KiCad' then gives
```
export.py:224: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--path", action="store", type="string", dest="path", default="~/", help="")
export.py:225: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option('-f', '--filetype', action='store', type='string', dest='filetype', default='jpeg', help='Exported file type')
export.py:226: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--crop", action="store", type="inkbool", dest="crop", default=False)
export.py:227: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--dpi", action="store", type="float", dest="dpi", default=600)
export.py:228: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--threshold", action="store", type="float", dest="threshold", default=128.0)
export.py:229: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openfactory", action="store", type="inkbool", dest="openfactory", default="true")
export.py:230: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--openkicad", action="store", type="inkbool", dest="openkicad", default="true")
export.py:231: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--autoflatten", action="store", type="inkbool", dest="autoflatten", default="true")
export.py:232: DeprecationWarning: Effect.OptionParser or `optparse` has been deprecated and replaced with `argparser`.You must change `self.OptionParser.add_option` to `self.arg_parser.add_argument`; the arguments are similar.
  self.OptionParser.add_option("--debug", action="store", type="inkbool", dest="debug", default=False)
export.py:748: DeprecationWarning: Effect.affect is now `Effect.run()`. The `output` argument has changed.
  e.affect()
export.py:347: DeprecationWarning: self.args[-1] is now self.options.input_file
  curfile = self.args[-1]
Traceback (most recent call last):
  File "export.py", line 752, in <module>
    _main()
  File "export.py", line 748, in _main
    e.affect()
  File "/usr/share/inkscape/extensions/inkex/deprecated.py", line 181, in affect
    return self.run(args=args)
  File "/usr/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 324, in effect
    self.processExportLayer()
  File "export.py", line 348, in processExportLayer
    layers = self.get_layers(curfile)
  File "export.py", line 548, in get_layers
    if  layer_label_name in self.layer_map.iterkeys():
AttributeError: 'dict' object has no attribute 'iterkeys'
```
and gives no output.

Sodium-Hydrogen, 2020-11-20:
What version of svg2shenzhen are you using? It looks like these error have been fixed since #76

f0x52, 2020-11-20:
oh my bad, I thought I was running an up-to-date version but, quite old actually :)

closed 2020-11-20 by f0x52