badgeek/svg2shenzhen #115: Trouble exporting to KiCad

pomd, 2020-12-01:
I'm encountering the following error using windows, inkscape v1.0.1,  svg2shenzhen 0.2.18.7

Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 497, in processExportLayer
    with open(kicad_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\David\\AppData\\Local\\Temp\\ink_ext_XXXXXX.pretty\\F.Cu_g840.kicad_mod'

jorticus, 2020-12-08:
I also encountered this, resolved it by upgrading from Inkscape 0.91 to 1.01

insomniacslk, 2020-12-18:
I am encountering this issue too, and already using inkscape 1.0.1. svg2shenzen latest version (commit hash 42378e5ab2fa74d7d0d03dff32bd50893117ce73 )

insomniacslk, 2020-12-18:
Well... that was easy. I am using the source distribution, so `bitmap2component` is not built for my platform. Build it with `src/build.sh` and copy the relevant file to the inkscape plugin location (e.g. `cp src/bitmap2component inkscape/svg2shenzen/bitmap2component_linux64`, see `.travis.yml` for other platforms).  This requires installing the dependencies (e.g. `potrace`).

Or, easier, download the binary distribution as specified in the README

nikhilrami, 2020-12-19:
I did all the process of copying and pasting three files on Inkscape. After restarting Inkscape, under Extensions drop down menu, I am getting svg2shenzen 0.2.18.7 option. But under this option 1.Prepare document 2.Export to kicad remains disabled. Only About option is active and on clicking on it, an About window opens up. I am using Inkscape verion 1.0.1(1.0.1+r74). Please help me out.

HackModBrew, 2021-01-25:
I am having an issue with exporting to kicad as well. I am using 2.18.7 and inkscape 1.0.2. When I hit Apply, nothing happens. No Kicad project or module is made.

iharshadec, 2021-01-25:
> I am having an issue with exporting to kicad as well. I am using 2.18.7 and inkscape 1.0.2. When I hit Apply, nothing happens. No Kicad project or module is made.

Can you share your file?

HackModBrew, 2021-01-26:
Absolutely! I appreciate you reaching out to help. I found a mistake I made
so tomorrow I will try it again and if it works I will let you know and
update my github comment to what I did wrong. If it still doesn't work I
will send my files to have you look at it.

Thanks,

Jon

hobipivo, 2021-01-31:
Export to KiCad does not work. There is no file in the registered folder D:\KiCad. I'm using Windows, inkscape v1.0.2 and svg2shenzhen 0.2.18.7. I have activated "Debug", but cannot find a debug report file.

Christian