badgeek/svg2shenzhen #116: Exported file ends up in "\AppData\Local\Temp\ink_ext_XXXXXX.pretty" rather than selected folder

robot-army, 2021-01-26:
Hi, 

On running export for the first time, and selecting a folder where I'd like the footprint to appear, I didn't find it where I selected. 

Enabling 'debug', the below output shows the file appearing in AppData\Local\Temp...

Other files appear in the selected folder (the .svg2shenzhen-cache\ etc is there). 

```
inkscape -C -d 1200 -o "[dialog selected folder]\.svg2shenzhen-cache\images\B.SilkS_g211.png" "[dialog selected folder]\.svg2shenzhen-cache\jog25ilt.svg"
"C:\Users\[username]\AppData\Roaming\inkscape\extensions\svg2shenzhen\bitmap2component.exe" "[dialog selected folder]\.svg2shenzhen-cache\images\B.SilkS_g211.png" "C:\Users\[username]\AppData\Local\Temp\ink_ext_XXXXXX.pretty\B.SilkS_g211.kicad_mod" B.SilkS true 1200 5
```

This is my first time trying this plugin (in recent memory at least!) so if I've set something up wrong, please let me know!

Thanks for all you do, I'm looking forward to getting a sweet logo in my silkscreen :)

iharshadec, 2021-01-27:
There is option where you want to export
[image]

benguru, 2021-02-09:
Same issue here. Did you find the solution to this ?

robot-army, 2021-02-09:
No, I didn't. @iharshadec is right, there is an option to tell it where to export, but on my install it doesn't seem to put it there. It appears instead in some /appdata/ windows thing I don't claim to understand at all. 

Glad to see it's not just me!

iharshadec, 2021-02-09:
> No, I didn't. @iharshadec is right, there is an option to tell it where to export, but on my install it doesn't seem to put it there. It appears instead in some /appdata/ windows thing I don't claim to understand at all.
> 
> Glad to see it's not just me!

What's your inkscape Verison? If you are other than 1.0 can you try to download and install Inskcape 1.0?