badgeek/svg2shenzhen #118: What you telling me?

Pipstapee, 2021-02-11:
Simple...What you telling me?

Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Users\Phil\AppData\Roaming\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 468, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 659, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range

Run the extension many times in Inkscape but now it's telling me the above.....What you telling me?
I'm not a coder so keep it simple. I only use Inkscape for one job and that's because there isn't a SVG2shenzhen for Photoshop.

iharshadec, 2021-02-12:
@Pipstapee is it possible to share your .svg file may be something on that itself is causing this.

Pipstapee, 2021-02-12:
I'm going to try a few things. I've used the Extension many times before.

Pipstapee, 2021-02-13:
[image]
Take a look mate. After reinstalling Inkscape and the extension it still doesnt write the files and shows the error log as above. I've just tried it with project previously compiled that worked before and it shows the same error log. To conclude, I shouldn't be so lazy and I should learn how to use professional software, feel free to Harshad to take a look and see what you find.

iharshadec, 2021-02-13:
This is JPEG file. I can not open this in my inkscape. Are this round "Holes". If you can share your .svg then I can try to export on my inkscape and see if any layer or anything else is creating problem or not.

iharshadec, 2021-02-13:
Here is my flow.
First prepare document.
[image]

[image]

Please find attached inkscape and exported file 
[Test.zip](https://github.com/badgeek/svg2shenzhen/files/5975547/Test.zip)

Pipstapee, 2021-02-13:
Done that buddy, hence me being here.

Pipstapee, 2021-02-13:
[Rings.zip](https://github.com/badgeek/svg2shenzhen/files/5975600/Rings.zip)
Knock yourself out buddy. From what I can see your just going down the normal user route and assuming I've not tried that.
It's got to be something else. I've installed Inkscape and extension on a laptop with only Gcode software on it and it says it's saved it but when you look in the save folder it's empty....

Pipstapee, 2021-02-13:
"This is JPEG file. I can not open this in my inkscape."
You can drag and drop Jpegs, what version of inkscape do you use?

Pipstapee, 2021-02-13:
I see you're using ver, 0.2.18.6 what version of Inkscape are you using?

iharshadec, 2021-02-13:
> I see you're using ver, 0.2.18.6 what version of Inkscape are you using?

I am using 1.0

Pipstapee, 2021-02-13:
So you couldn't possible direct me to a solution because we're both using different versions.

Pipstapee, 2021-02-13:
Inkscape 1.0.2 svg2 shen 0.2.18.7
This is what I've tried:
Earlier versions of each as well as 32bit and 64bit.
Putting extension in Inkscape extension and user extension folder, all of the above don't work.
The reason for raising this issue in the first place is that I've been running Ink v1.0.2 S2S 0.2.18.7 for about 2 months with no issues, I started using it again a few days ago and then the above error happened...
If you need me to elaborate please let me know, I may not be a coder but don't assume I'm a simpleton.