badgeek/svg2shenzhen #119: Unable to export a single rectangular edge cut

cutlasses, 2021-02-14:
I'm using Inkscape for Mac 1.0.2. Since upgrading I have been unable to export using SVG 2 Shenzhen. I have the most up to date version (0.2.18.7).

Steps taken.

1. svg2shenzhen -> Prepare Document (leave options as default)
2. Draw a rectangle on the document
3. Convert rectangle Object to Path
4. svg2shenzhen -> Export to KiCad
5. Export fails with the following error

`Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 468, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 659, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range
`

If I leave out step 3 (converting to a path), the export works, but I don't see anything when the project is loaded in KiCad.

Thanks!

cutlasses, 2021-02-14:
I have discovered that, if, after converting the rectangle to a path, I also apply Path->Union, the error goes away and the rectangle appears in KiCad! But, I am unable to get circular cuts to work in this way, so am still unable to achieve what I need to do.

dandelany, 2021-04-25:
I am also experiencing this issue when attempting to export the Edge Cuts layer, running svg2shenzhen 0.2.18.7 and Inkscape 1.0.1 on Windows 10.  I have an Edge Cuts layer with bezier curves and exporting it yields the error:
```
Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 468, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 659, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range
```
I can also confirm that if I apply Path -> Union the error goes away, but the exported KiCad file does not accurately capture the curves:

[image]

[image]