badgeek/svg2shenzhen #120: No perfect horizontal and vertical lines

BitBlit88, 2021-02-16:
I observed that some lines that were perfect horizontal or vertical in Inkscape aren't perfect horizontal or vertical after exporting them to Kicad.
I also noticed that it is depending on the positioning. While a line is not perfect horizontal or vertical, it will be if it's moved some mm in x or y position.
It looks like a rounding error.
I attached some screenshots, the source svg file and the generated Kicad mod.

[image]
[image]
[antenna.zip](https://github.com/badgeek/svg2shenzhen/files/5988523/antenna.zip)

Sodium-Hydrogen, 2021-02-26:
This is an artifact from the lossy conversion that svg2shenzhen uses. See #107 
I am working on making [svg2mod](/svg2mod/svg2mod) a more complete tool then I'll be able to work on adding it as an optional parser. If you are comfortable with the command line svg2mod can work for your case until then

BitBlit88, 2021-03-03:
thank you, svg2mod worked perfectly.

closed 2021-03-03 by BitBlit88