badgeek/svg2shenzhen #125: KiCAD Module not created

wraybowling, 2021-04-24:
Others have pointed out an error when trying to export. I am experiencing similar issues. Here you can see it even applies to the example file provided in the readme. As much information as I could think to include also provided below. I would guess that this isn't a permissions problem as the script is already creating plenty of other files just fine (see screenshot). 

svg2shenzhen: 0.2.18.7
OS: Linux, Ubuntu 20.04.2, Budgie
Inkscape: 1.0.2 (installed via snap)
python --version: Python 2.7.18
python3 --version: Python 3.8.5
File used: [Example PCB](https://raw.githubusercontent.com/badgeek/svg2shenzhen-next/master/examples/viruspcb.svg)

Error:
```
Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "/snap/inkscape/8746/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 497, in processExportLayer
    with open(kicad_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/wray/Downloads/export-example/.svg2shenzhen-cache/viruspcb.pretty/F.Cu_g8165.kicad_mod'
```

[image]

wraybowling, 2021-04-24:
bitmap2component binaries are in fact set as executable for all roles, so that's not the issue.
```
drwxr-xr-x 2 wray wray    4096 Aug 22  2020 .
drwxr-xr-x 3 wray wray    4096 Apr 24 15:50 ..
-rwxr-xr-x 1 wray wray 1448030 Aug 22  2020 bitmap2component.exe
-rwxr-xr-x 1 wray wray  755571 Aug 22  2020 bitmap2component_linux64
-rwxr-xr-x 1 wray wray  927400 Aug 22  2020 bitmap2component_osx
-rw-r--r-- 1 wray wray   25218 Aug 22  2020 export.py
-rw-r--r-- 1 wray wray   19801 Aug 22  2020 prepare.py
```

iharshadec, 2021-04-25:
As I mentioned in other issue. Can you please try once by installing older version of Inkscape 1.0 instead of 1.0.2?