badgeek/svg2shenzhen #127: Export failure

endstreet, 2021-05-30:
Traceback (most recent call last):
  File "export.py", line 750, in <module>
    _main()
  File "export.py", line 746, in _main
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "export.py", line 323, in effect
    self.processExportLayer()
  File "export.py", line 468, in processExportLayer
    kicad_edgecut_string = self.exportEdgeCut(kicad_mod = options.filetype == "kicad_module")
  File "export.py", line 659, in exportEdgeCut
    y = (path[1][1])
IndexError: list index out of range

[PouchConn.zip](https://github.com/badgeek/svg2shenzhen/files/6565920/PouchConn.zip)

I have uploaded the .SVG file.

AEFeinstein, 2021-09-01:
I'm experiencing this as well

AEFeinstein, 2021-09-01:
The export works with Inkscape 1.1 (c68e22c387, 2021-05-23)