badgeek/svg2shenzhen #133: Can I get some help, Non of the KiCad files being made on export.

PigeonFX, 2021-11-02:
SO, I had it working then updated to Inkscape to 1.1.1 and everything broke. I installed and reinstalled inkscape 1.1, but now its not exporting to KiKad.

I used the example PCB file viruspcb linked in the tutorial.  use the export function with open KiCad, KiCad doesn't open and in the selected folder there are no KiCad files. 

[image]

[image]

[image]

rcludwick, 2021-11-02:
At this point I'm thinking the best approach until they fix this would be to figure out how to export each layer into a dxf, and then import it back into kicad.

I'm thinking this may be abandonware given the number of issues vs pull requests over the last year.

PigeonFX, 2021-11-02:
> That said you might try inkscape 1.0 since that seems to be the only version I've seen it works with.

Same problem with 1.0, 1.0.1, 1.0.2, 1.1 and 1.1.1 

The frustrating part is I had it working yesterday on 1.1 (should never just left it alone).  

I Will look into the .dxf option, I do all the artwork in illustrator so getting dfx is not a problem, but bringing it into KiCad is going to take some reading.

rcludwick, 2021-11-02:
So you can import DXF graphics into any layer into kicad_pcb.   Though I believe solder mask has to be inverted (in other words the positive area is where the solder mask is removed)

If I were going to do it, I would make a custom footprint with the copper, solder mask, and silkscreen dxf layers, then import the edge-cut layer directly into kicad_pcb.   

You can make a registration mark on all layers to make it easy to line up, like a small "X" somewhere.

PigeonFX, 2021-11-02:
> 
> If I were going to do it, I would make a custom footprint with the copper, solder mask, and silkscreen dxf layers, then import the edge-cut layer directly into kicad_pcb.
> 

How about drill holes, will KiCad be OK with them as dfx or will it just see them as small pockets to rout out?

rcludwick, 2021-11-02:
For drills I would make a series of marks for the center holes directly in
maybe the silkscreen layer, and then just manually add them in the
footprint editor.

For routed holes I would use the edge cut layer.

PigeonFX, 2021-11-02:
Well I have a little progress the extension seems to be making all the layers to be imported:
[image]

Just still not making the .kicad_pcb, .pro  or Pcbnew files

[image]

[image]

rcludwick, 2021-11-02:
I was looking at svg2mod from here:

https://github.com/svg2mod/svg2mod

I'm going to try that this evening.  

It would be similar in that certain layers need to be named the same way, but the last release was back in april.

PigeonFX, 2021-11-02:
> It would be similar in that certain layers need to be named the same way, but the last release was back in April.

Recognise that name from them helping people on here, looks like they decided to do it themselves! 

Please let me know how you get on, just been trying with dfx files, but ran into the issue that not all the copper traces I make are even weight (I trace a lot of old 1960's PCBS that were hand-drawn).

Would love to hear how you get on with it!

PigeonFX, 2021-11-02:
For what its worth.....Got it running! 

Anyone who runs into the same issue Select the Export As: Project option. 

[image]

I also added drayde files that mean you can have non plated through holes (that is going to save me hours of clicking!) Thank you drayde, if you see this! 

https://github.com/drayde/svg2shenzhen/tree/master/inkscape

[image]
[image]

rcludwick, 2021-11-02:
I think svg2mod is easier.  It's not perfect, but it seems quite a bit easier to use.