badgeek/svg2shenzhen #134: Prepare has no attribute 'run'

pierstu, 2021-12-05:
Hello, i'm likely having a python issue.

Bunsenlabs / Debian Buster, Inkscape 0.92.4 installed from apt
When I try to setup a new document, after hitting Apply i have a modal showing this error msg :

> Traceback (most recent call last):
  File "prepare.py", line 322, in <module>
    _main()
  File "prepare.py", line 318, in _main
    e.run()
AttributeError: Svg2ShenzhenPrepare instance has no attribute 'run'

Here's what python -tt returns in the terminal :

> $ python -tt /usr/share/inkscape/extensions/svg2shenzhen/prepare.py
Traceback (most recent call last):
  File "/usr/share/inkscape/extensions/svg2shenzhen/prepare.py", line 322, in <module>
    _main()
  File "/usr/share/inkscape/extensions/svg2shenzhen/prepare.py", line 318, in _main
    e.run()
AttributeError: Svg2ShenzhenPrepare instance has no attribute 'run'

Of course i've never been taught python and would like to bypass this error. Did anyone find a workaround?
Thanks, have a nice day

Sodium-Hydrogen, 2021-12-05:
From the Readme

> Warning: starting from 0.2.18 svg2shenzhen only support Inkscape 1.0 and above

pierstu, 2021-12-06:
Tried with Inkscape 1.0.0 Appimage
Now i got this error, still at validation :

> Fatal Python error: initfsencoding: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Looks like i have only Python 2.7 on this machine.
I'll try with another session later this week.

Thanks