badgeek/svg2shenzhen #142: Incompatable with inkscape 1.3

CountMurphy, 2023-08-18:
Using inkscape 1.3, all of the .kicad_mod files produced by this plugin are 0 bytes. The only layer that is being translated correctly are edge cuts. Everything else is completely omitted from the export.

marhalloweenvt, 2023-09-01:
I am also facing same problem and have to downgraded to 1.2 to work with it for now.

CountMurphy, 2023-09-05:
Since it looks like this project might be dead, I moved to this command line tool: https://github.com/mtl/svg2mod

it works out of the box with almost all of my board projects

Sodium-Hydrogen, 2023-09-05:
The version of svg2mod you linked is abandoned. 

There is a newer and more updated version at:
https://github.com/svg2mod/svg2mod

CountMurphy, 2023-09-05:
My mistake. Thanks for the updated link!

badgeek, 2023-09-06:
> Since it looks like this project might be dead, I moved to this command line tool: https://github.com/mtl/svg2mod
> 
> it works out of the box with almost all of my board projects

sorry for not having time to update this incompability. recently and currently struggling with my wife having diagnozed with cancer stadium 3B

AEFeinstein, 2023-09-06:
My condolences. I hope your wife beats cancer into remission. Take as much time with her as you can get; this project isn't important in comparison.

CountMurphy, 2023-09-06:
So sorry to hear about your wife. Please do not stress about this project. 100% agree with AEFeinstein.

badgeek, 2024-03-21:
https://github.com/badgeek/svg2shenzhen/issues/144