badgeek/svg2shenzhen #144: goodbye, this project is discontinued

badgeek, 2024-03-21:
It is with a heavy heart that I write this message. Due to the recent passing of my beloved wife, I find myself in a place where continuing the development and maintenance of svg2shenzhen is beyond my current emotional capacity. Her loss has deeply affected me, and as a result

I've lost the motivation necessary to push this project forward. I will archive the project, and I would be happy if someone were to fork it and continue its development. However, I also believe there are many better tools out there that have rendered svg2shenzhen obsolete.

I am proud of how this project has enabled many people to create beautiful PCBs. I had hoped to collaborate with my wife on her beautiful illustrations, but sadly, she will never see it.

attached below link to winterbloom as an alternative to create pcb art and works better than svg2shenzhen

https://github.com/wntrblm/Gingerbread#using-trace

AEFeinstein, 2024-03-21:
I'm terribly sorry to hear that. She sounded like a wonderful wife. I hope you find peace.

patfelst, 2024-03-21:
take care @badgeek I'm sorry for your loss